cyberbotics/webots_ros2 #306: Error using URDF Import

HeChengHui, 2021-10-13:
I managed to install ROS2-Foxy and webots 2021b and was able to get 'ros2 launch webots_ros2_demos armed_robots.launch.py' working.

However, when I try to convert URDF to proto in (https://github.com/cyberbotics/webots_ros2/wiki/Using-URDF-or-Xacro), an error occurs as shown:
[image]

I have sourced both webots_ros2 and ros2 using their `local_setup.bat` before converting.

BenjaminHug8, 2021-10-14:
If you tried to convert URDF to proto with the command found on the wiki page, it uses "ur5e_robot.urdf" that is also used in armed_robots.launch.py, so regarding the error did you tried to convert while the demo was running ?

HeChengHui, 2021-10-14:
@BenjaminHug8 No I wasn't running any demo at that time. I have also tried it just now, without any use of ROS2 beforehand, and it gave the same error but a different URDF file this time.
[image]

Full console:
[image]

BenjaminHug8, 2021-10-15:
On my side the function is working with "webots_ur5e_description.urdf" for example on linux.

```
benjamin@benjamin-computer:~/ros2_ws/src/webots_ros2$ ros2 run webots_ros2_importer urdf2proto   --input=webots_ros2_universal_robot/resource/webots_ur5e_description.urdf   --disable-mesh-optimization   --output=ur5e_robot.proto
There are 0 links, 0 joints and 0 sensors

```
Can you try to go in your `AppData\Local\Temp` and remove the files ending by ".urdf" and rerun the command ?

HeChengHui, 2021-10-16:
@BenjaminHug8  I did try that out the previous time and it gave me the same error but for a different urdf file. The error also appears when I cleared all the urdf files in that Temp folder. 
I have since managed to get my urdf file converted using an edited urdf2webots package, so I wont be using this anymore. I suspect it might be due to Windows because it worked on my friend's Linux system.