cyberbotics/webots_ros2 #338: How to

Ryoma-Liu, 2021-11-15:


BenjaminHug8, 2021-11-15:
> When I try to generate an interface for my webots world simulation. I did not see that we need first write a controller and set it to extern, but when I run "ros2 run webots_ros2_driver driver", an error occurs and says that no available robot controller found. 

To run `ros2 run webots_ros2_driver driver` you need to have a world with a robot having its controller set to `<extern>`, as this command will search for a robot having an external controller. Otherwise you will get this error as `driver` from `webots_ros2_driver` will try to control the robot without being allowed to do it.

> when I run this command, where it will generate the sample code structure?

I am not sure about what you mean by this sentence, is it that you wonder if the command create a controller file as the standard Webots controller you can write in C, C++, Python ... ?

In that case we could say that the controller is already written and it is `driver` but in fact it is more an interface and you are suppose to write a plugin either in [C++](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Creating-a-Custom-Plugin) or in [Python](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Creating-a-Custom-Python-Plugin).

I invite you to check also the other [tutorials](https://github.com/cyberbotics/webots_ros2/wiki/Tutorials) and you might take a look at the [mavic](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_mavic) or [tesla](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_tesla) examples to have an idea how to create plugins in Python (tesla example also has an extra ROS node).

Ryoma-Liu, 2021-11-15:
Thanks for your explanation. Sorry for my misunderstanding of the code structure. I just read the source code of the tesla example and have a better understanding of the workflow, and try to make an initial trial for the python plugin. I replace the world file with my own world file and close the controller plugin.  It opens my world, and then exits, and reports the following erros:
[image]

Below is my launch file. 
`"""Launch Webots Tesla driver."""

import os
import pathlib
from launch.substitutions import LaunchConfiguration
from launch.actions import DeclareLaunchArgument
from launch.substitutions.path_join_substitution import PathJoinSubstitution
from launch import LaunchDescription
from ament_index_python.packages import get_package_share_directory
from launch_ros.actions import Node
from webots_ros2_driver.webots_launcher import WebotsLauncher

def generate_launch_description():
    package_dir = get_package_share_directory('webots_ros2_tesla')
    world = LaunchConfiguration('world')
    robot_description = pathlib.Path(os.path.join(package_dir, 'resource', 'tesla_webots.urdf')).read_text()

    webots = WebotsLauncher(
        world=PathJoinSubstitution([package_dir, 'worlds', world])
    )

    tesla_driver = Node(
        package='webots_ros2_driver',
        executable='driver',
        output='screen',
        parameters=[
            {'robot_description': robot_description},
        ]
    )

    lane_follower = Node(
        package='webots_ros2_tesla',
        executable='lane_follower',
    )

    return LaunchDescription([
        DeclareLaunchArgument(
            'world',
            default_value='agiletaur.wbt',
            description='Choose one of the world files from `/webots_ros2_tesla/worlds` directory'
        ),
        webots,
        tesla_driver,
    ])
`
When I open the simulator， and run `ros2 run webots_ros2_driver driver`， it reports
[image]

Is the reason that I use the wrong world?

BenjaminHug8, 2021-11-15:
What we can get from your error is a bad allocation, hard to tell you more without more information. Is your world file at the same place that the original tesla world file is (in the same folder)?

To execute the launch file of tesla did you run `ros2 launch webots_ros2_tesla robot_launch.py` (see [here](https://github.com/cyberbotics/webots_ros2/wiki/Example-Tesla-Model-3))?

> When I open the simulator， and run `ros2 run webots_ros2_driver driver` ， it reports [image]
> 
> Is the reason that I use the wrong world?

`ros2 run webots_ros2_driver driver` without a `robot_description` (containing an urdf file) will just use default parameters, that is why you get this report.

Ryoma-Liu, 2021-11-17:
Thanks for your explanation. When I am trying to make a c++ plugin. It reports the following error,

lukicdarkoo, 2021-11-18:
@Ryoma-Liu Can share the code of your plugin? You are building the plugin for Galactic (assumption based on the fact that FastRTPS is not available)?

Ryoma-Liu, 2021-11-22:
THANKS. I did get the code work. But I am quited fused about the project structure and the use of each part. 
As you see, there are many folder in this repo
[image]
If my goal is to transfer my previous c++ controller code and a legged robot I built in webots to ros2 interface. Which package does I need？ Only the webots_ros2_driver?  How can I transfer my code to the ros2?  It would be great that you can tell me where I can know the webots information I can get?
Thank  you very much.

lukicdarkoo, 2021-11-23:
You only need the `webots_ros2_driver` package and maybe `webots_ros2_control` (if you want to use `ros2_control` to control the motors). However, you should not worry about the project structure as you can install them through apt, `sudo apt install ros-foxy-webots-ros2-driver os-foxy-webots-ros2-control`

Ryoma-Liu, 2021-12-16:
> You only need the `webots_ros2_driver` package (that depend on `webots_ros2_msgs`) and maybe `webots_ros2_control` (if you want to use `ros2_control` to control the motors). However, you should not worry about the project structure as you can install the packages through apt:
> 
> ```
> sudo apt install ros-foxy-webots-ros2-driver ros-foxy-webots-ros2-control
> ```
> 
> We have a lot of examples that simply demonstrate Webots and ROS 2 usage:
> 
> * webots_ros2_epuck
> * webots_ros2_mavic
> * webots_ros2_tesla
> * webots_ros2_tiago
> * webots_ros2_turtlebot
> * webots_ros2_universal_robot

What's the usage of webots ros2 control package?  Can I publish a velocity command, and then the robot in the webot could move?

lukicdarkoo, 2021-12-16:
> What's the usage of webots ros2 control package? Can I publish a velocity command, and then the robot in the webot could move?

Yes, you also get Odometry and TF2 messages. But that is just one controller, you can load any of the following:
https://github.com/ros-controls/ros2_controllers
and more will be added.

lukicdarkoo, 2021-12-16:
I will close the issue as it seems to be resolved. Feel free to open it if needed.

For non-bug related questions please use:
- GitHub Discussions: https://github.com/cyberbotics/webots/discussions
- Discord: https://discordapp.com/invite/nTWbN9m

closed 2021-12-16 by lukicdarkoo