cyberbotics/webots_ros2 #343: TurtleBot3 simulation doesn't work at 50Hz as specified

lukicdarkoo, 2021-11-24:
Given:
- https://github.com/cyberbotics/webots_ros2/blob/c1b3a055f9b3d931066083e561c3dbee5b386c3a/webots_ros2_turtlebot/resource/ros2control.yml#L3
- https://github.com/cyberbotics/webots_ros2/blob/c1b3a055f9b3d931066083e561c3dbee5b386c3a/webots_ros2_turtlebot/resource/ros2control.yml#L19

We should receive Odometry messages at 50Hz, but instead, we get them at 30Hz.

This is reasonable as a simulation with a `basicTimeStep` at 32ms cannot deliver that high frequency. However, reducing the `basicTimeStep` increases the frequency to 60Hz which is wrong.

This indicates there is a bug in `webots_ros2_control`.

BenjaminHug8, 2021-12-03:
Waiting on [this `ros2_controllers` issue](https://github.com/ros-controls/ros2_controllers/issues/274) (I asked if we can add a `publish_rate` as it exist in ROS 1 but not in ROS 2).

lukicdarkoo, 2021-12-03:
I believe a bigger issue is that the `update_rate` param is ignored. We are missing this part:
https://github.com/ros-simulation/gazebo_ros2_control/blob/519b0066aa36bad919d0db94f65e343fcfe00b3e/gazebo_ros2_control/src/gazebo_ros2_control_plugin.cpp#L367-L375

Another issue is that Webots typically operates at a low frequency of 31Hz (32ms) and the typical control update rate is 50Hz. Therefore, with the current implementation, we cannot deliver 50Hz with timestep 32ms. Even if we reduce the timestep in the simulation to like 16ms the update will not be accurate. Other physics engines operate with much lower timesteps and it is easier to solve.

However, I am not sure what it a good way to solve it in our case.

PR #346 (2021-12-16): Fix controller_manager update rate
Fixes #343

There is not much we can do to tackle this. Basically:
- If basicTimeStep doesn't match the update rate then we notify the user about that.
- If a user uses low basicTimeStep then we skip some updates.

**Tasks**
- [x] Fix the update_rate
- [x] Fix copyright 
- [x] Fix CI
closed 2021-12-16 by lukicdarkoo