cyberbotics/webots_ros2 #344: Got Link error when build libwebots_ros2_control_system.dylib on macOS for foxy

zegangYang, 2021-12-05:
**Describe the Bug**
Got Link error when build libwebots_ros2_control_system.dylib on macOS for foxy

**Expected behavior**
build pass

**Affected Packages**
  - webots_ros2_control
  - webots_ros2_driver

**build error log**
```
[ 75%] Linking CXX shared library libwebots_ros2_control_system.dylib
Undefined symbols for architecture x86_64:
  "webots::Robot::getMotor(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      webots_ros2_control::Ros2ControlSystem::init(webots_ros2_driver::WebotsNode*, hardware_interface::HardwareInfo const&) in Ros2ControlSystem.cpp.o
```
**System**
 - Webots Version: 2021b
 - ROS Version: foxy
 - Operating System: macOS Catalina
 - Graphics Card: 

**Quick fix**

change `webots_ros2/webots_ros2_control/CMakeLists.txt`
append below link libraries to line 70
```
set(WEBOTS_DRIVER_DARWIN "build_root/install/webots_ros2_driver/lib")
target_link_libraries(
  ${PROJECT_NAME}_system
  ${WEBOTS_DRIVER_DARWIN}/libController.dylib
  ${WEBOTS_DRIVER_DARWIN}/libCppCar.dylib
  ${WEBOTS_DRIVER_DARWIN}/libCppController.dylib
  ${WEBOTS_DRIVER_DARWIN}/libCppDriver.dylib
  ${WEBOTS_DRIVER_DARWIN}/libcar.dylib
  ${WEBOTS_DRIVER_DARWIN}/libdriver.dylib
)
```

If there is a better way to fix this ? Thanks a lot

lukicdarkoo, 2021-12-10:
Thank you for the investigation and proposed solutions! It is true that we stopped testing on macOS after the macOS support had dropped to Tier 3. 

Could you please make a PR with the proposed changes?

zegangYang, 2021-12-11:
> Thank you for the investigation and proposed solutions! It is true that we stopped testing on macOS after the macOS support had dropped to Tier 3.
> 
> Could you please make a PR with the proposed changes?

Of course

PR #348 (2022-01-03): fix: 🐛 link error when build libwebots_ros2_control on macOS
fix link error when build libwebots_ros2_control_system.dylib on macOS
ros2 version : foxy
operating system : macOS catalina

✅ Closes: #344

**Description**
fix Link error when build libwebots_ros2_control_system.dylib on macOS

**Related Issues**
This pull-request fixes issue #344 

**Affected Packages**
List of affected packages:
  - webots_ros2_controller

**Additional context**
Maybe there have some issues to build webs_ros2 package on macOS Catalina if you install ros2 from binary package,
here is solutions:
```
rosinstall_gene [... cut]
commit 5de2db72 (2022-01-03): fix: 🐛 link error when build libwebots_ros2_control on macOS (#348)

* fix: 🐛 link error when build libwebots_ros2_control on macOS

fix Link error when build libwebots_ros2_control_system.dylib on macOS

✅ Closes: #344

* Simplify the linking

* Update CMakeLists.txt

* Update CMakeLists.txt
closed 2022-01-03 by zegangYang