cyberbotics/webots_ros2 #350: executable 'spawner.py' not found on the libexec directory controller_manager

dsauval, 2021-12-15:
**Describe the Bug**
When trying to ros2 launch the getting started launchfile, I get an error about spawner.py that can't be find :
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at C:\dev\ros2_foxy\Lib\site-packages\launch\launch_service.py:226> exception=SubstitutionFailure("executable 'spawner.py' not found on the libexec directory 'C:\\Users\\dsauval\\webots_ws\\install\\controller_manager\\lib\\controller_manager' ")>

Even though I can see myself the file spawner.py at 'C:\\Users\\dsauval\\webots_ws\\install\\controller_manager\\lib\\controller_manager'.

If I run the command : ros2 run controller_manager spawner.py "test", it works, I get the expected output "[INFO] [1639580496.533760400] [spawner_test]: Waiting for /controller_manager services". I tried a minimal launchfile that is the equivalent of ros2 run controller_manager spawner.py "test", but I get the spawner.py not found error.

I am not sure why it would work using ros2 run, but not using ros2 launch. My understanding would be that both commands use the same environment variables..

**Steps to Reproduce**
1. Create a new workspace : C:\Users\dsauval\webots_ws
2. Go to : https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install, and follow the instructions for Windows.
(my cmd could not find vcs, I had to pip install vcstool, maybe it should be added to pip install rosinstall_generator colcon-common-extensions lark)
3. ros2 launch webots_ros2_universal_robot multirobot_launch.py or ros2 launch webots_ros2_universal_robot robot_launch.py (same error for both)
4. See error.

**Expected behavior**
A clear and concise description of what you expected to happen.

**System**
 - Webots Version: R2021b. Installed using the binaries.
 - ROS Version: ROS2 Foxy 
 - Operating System: Windows 10 version 2004
 - Graphics Card: Nvidia RTX 2060

Feel free to ask for additional details ;) . Thank you !

lukicdarkoo, 2021-12-15:
We have the same problem with macOS

dsauval, 2021-12-16:
I guess we have to wait for ros2_control to release a fixed versoin right ?

lukicdarkoo, 2021-12-20:
We should be able to fix on our side as well

lukicdarkoo, 2021-12-20:
@dsauval I have just tested and I cannot reproduce the problem:
```
C:\Users\lukic\ros2_ws>ros2 launch webots_ros2_universal_robot robot_launch.py
[INFO] [launch]: All log files can be found below C:\Users\lukic\.ros/log\2021-12-20-10-30-21-596848-DESKTOP-035VG4O-408
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING: Target Webots version `R2021b` is not found, fallback to `R2021b revision 1`
[INFO] [spawner.py-1]: process started with pid [2916]
[INFO] [spawner.py-2]: process started with pid [5412]
[INFO] [webots-3]: process started with pid [8536]
[INFO] [robot_state_publisher.EXE-4]: process started with pid [10588]
[INFO] [driver.EXE-5]: process started with pid [11284]
[driver.EXE-5] Cannot open \\.\pipe\webots_1384_912649329. Retrying in 1 second.
[driver.EXE-5] Delete C:\Users\lukic\AppData\Local\Temp/webots-1384 to clear this warning.
[robot_state_publisher.EXE-4] Parsing robot urdf xml string.
[robot_state_publisher.EXE-4] [INFO] [1639992622.008101700] [robot_state_publisher]: got segment
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 2)
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 3)
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 4)
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 5)
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 6)
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 7)
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 8)
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 9)
[driver.EXE-5] Cannot open file: C:\Users\lukic\AppData\Local\Temp/webots-12316/WEBOTS_SERVER (retry count 10)
[driver.EXE-5] Error: The specified module could not be found.
[driver.EXE-5]  (dynamic library)
[driver.EXE-5] Error: failed to load C:/Program Files/Webots/resources/projects/plugins/robot_windows/generic/generic.dll library
[driver.EXE-5] [WARN] [1639992632.241711900] [UR5e]: Desired controller update period (50ms / 20Hz) is different from the Webots timestep (16ms). Please adjust the `update_rate` parameter in the `controller_manager` or the `basicTimeStep` parameter in the Webots `WorldInfo` node.
[robot_state_publisher.EXE-4] Parsing robot urdf xml string.
[robot_state_publisher.EXE-4] Link shoulder_link had 1 children
[robot_state_publisher.EXE-4] Link upper_arm_link had 1 children
[robot_state_publisher.EXE-4] Link forearm_link had 1 children
[robot_state_publisher.EXE-4] Link wrist_1_link had 1 children
[robot_state_publisher.EXE-4] Link wrist_2_link had 1 children
[robot_state_publisher.EXE-4] Link wrist_3_link had 2 children
[robot_state_publisher.EXE-4] Link ROBOTIQ 3f Gripper had 3 children
[robot_state_publisher.EXE-4] Link finger_middle_link_1 had 1 children
[robot_state_publisher.EXE-4] Link finger_middle_link_2 had 1 children
[robot_state_publisher.EXE-4] Link finger_middle_link_3 had 0 children
[robot_state_publisher.EXE-4] Link finger_1_link_0 had 1 children
[robot_state_publisher.EXE-4] Link finger_1_link_1 had 1 children
[robot_state_publisher.EXE-4] Link finger_1_link_2 had 1 children
[robot_state_publisher.EXE-4] Link finger_1_link_3 had 0 children
[robot_state_publisher.EXE-4] Link finger_2_link_0 had 1 children
[robot_state_publisher.EXE-4] Link finger_2_link_1 had 1 children
[robot_state_publisher.EXE-4] Link finger_2_link_2 had 1 children
[robot_state_publisher.EXE-4] Link finger_2_link_3 had 0 children
[robot_state_publisher.EXE-4] Link distance sensor had 0 children
[robot_state_publisher.EXE-4] [INFO] [1639992632.328999600] [robot_state_publisher]: got segment ROBOTIQ 3f Gripper
[robot_state_publisher.EXE-4] [INFO] [1639992632.329049900] [robot_state_publisher]: got segment base_link
[robot_state_publisher.EXE-4] [INFO] [1639992632.329066500] [robot_state_publisher]: got segment distance sensor
[robot_state_publisher.EXE-4] [INFO] [1639992632.329080300] [robot_state_publisher]: got segment finger_1_link_0
[robot_state_publisher.EXE-4] [INFO] [1639992632.329093100] [robot_state_publisher]: got segment finger_1_link_1
[robot_state_publisher.EXE-4] [INFO] [1639992632.329105200] [robot_state_publisher]: got segment finger_1_link_2
[robot_state_publisher.EXE-4] [INFO] [1639992632.329118000] [robot_state_publisher]: got segment finger_1_link_3
[robot_state_publisher.EXE-4] [INFO] [1639992632.329131000] [robot_state_publisher]: got segment finger_2_link_0
[robot_state_publisher.EXE-4] [INFO] [1639992632.329143500] [robot_state_publisher]: got segment finger_2_link_1
[robot_state_publisher.EXE-4] [INFO] [1639992632.329155800] [robot_state_publisher]: got segment finger_2_link_2
[robot_state_publisher.EXE-4] [INFO] [1639992632.329171700] [robot_state_publisher]: got segment finger_2_link_3
[robot_state_publisher.EXE-4] [INFO] [1639992632.329183800] [robot_state_publisher]: got segment finger_middle_link_1
[robot_state_publisher.EXE-4] [INFO] [1639992632.329195600] [robot_state_publisher]: got segment finger_middle_link_2
[robot_state_publisher.EXE-4] [INFO] [1639992632.329207000] [robot_state_publisher]: got segment finger_middle_link_3
[... cut ...]
[spawner.py-2] [INFO] [1639992622.589807300] [spawner_ur_joint_trajectory_controller]: Waiting for /controller_manager services
```
I cleaned my workspace and followed [the instructions](https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#windows), but I cannot reproduce the spawner issue. I have ROS Foxy installed according to [these instructions](https://docs.ros.org/en/foxy/Installation/Windows-Install-Binary.html).

How did you install the ros2_control dependency? Did you manually clone it or you used rosdep to take care of it?

DannySauval, 2021-12-20:
Hi, thank you for the feedback.
I just tried to build everything on my personal computer and it works like a charm ! Meaning I have some issue on the other computer, but this is my problem ;p I will update the issue with the solution when I find the solution.

Thank you for the help !

dsauval, 2021-12-20:
Hi, thank you for the feedback.

I just tried to build everything on my personal computer and it works like a charm meaning I have some issue on the other computer, but this is my problem ;p I will update the issue with the solution when I find the solution.

Thank you for the help

closed 2021-12-20 by dsauval