cyberbotics/webots_ros2 #358: Python plugins are not working on macOS

lukicdarkoo, 2021-12-21:
Here is a backtrace from `lldb`:
```
Process 13370 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x8)
    frame #0: 0x0000000140f06bf1 Python`PyTuple_New + 241
Python`PyTuple_New:
->  0x140f06bf1 <+241>: movq   0x8(%rcx), %rdx
    0x140f06bf5 <+245>: movq   %rax, (%rdx)
    0x140f06bf8 <+248>: movq   -0x8(%r14), %rsi
    0x140f06bfc <+252>: andq   $0x3, %rsi
Target 0: (driver) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x8)
  * frame #0: 0x0000000140f06bf1 Python`PyTuple_New + 241
    frame #1: 0x0000000140f097d9 Python`PyType_Ready + 233
    frame #2: 0x0000000140f09795 Python`PyType_Ready + 165
    frame #3: 0x0000000140e8872f _vehicle.so`SwigPyPacked_TypeOnce() + 143
    frame #4: 0x0000000140e81dff _vehicle.so`SwigPyPacked_type() + 47
    frame #5: 0x0000000140e81922 _vehicle.so`PyInit__vehicle + 34
    frame #6: 0x0000000100b0a432 Python`_PyImport_LoadDynamicModuleWithSpec + 632
    frame #7: 0x0000000100b09db5 Python`_imp_create_dynamic + 291
    frame #8: 0x0000000100a7d7f1 Python`cfunction_vectorcall_FASTCALL + 169
    frame #9: 0x0000000100a4f333 Python`PyVectorcall_Call + 108
    frame #10: 0x0000000100aecaeb Python`_PyEval_EvalFrameDefault + 28754
    frame #11: 0x0000000100aefd09 Python`_PyEval_EvalCodeWithName + 1952
    frame #12: 0x0000000100a4fa51 Python`_PyFunction_Vectorcall + 227
    frame #13: 0x0000000100aef1e6 Python`call_function + 346
    frame #14: 0x0000000100aec5c9 Python`_PyEval_EvalFrameDefault + 27440
    frame #15: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #16: 0x0000000100aef1e6 Python`call_function + 346
    frame #17: 0x0000000100aec5a9 Python`_PyEval_EvalFrameDefault + 27408
    frame #18: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #19: 0x0000000100aef1e6 Python`call_function + 346
    frame #20: 0x0000000100aec671 Python`_PyEval_EvalFrameDefault + 27608
    frame #21: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #22: 0x0000000100aef1e6 Python`call_function + 346
    frame #23: 0x0000000100aec671 Python`_PyEval_EvalFrameDefault + 27608
    frame #24: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #25: 0x0000000100aef1e6 Python`call_function + 346
    frame #26: 0x0000000100aec671 Python`_PyEval_EvalFrameDefault + 27608
    frame #27: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #28: 0x0000000100a50d37 Python`object_vacall + 277
    frame #29: 0x0000000100a50e78 Python`_PyObject_CallMethodIdObjArgs + 217
    frame #30: 0x0000000100b08ef9 Python`PyImport_ImportModuleLevelObject + 1564
    frame #31: 0x0000000100aeac95 Python`_PyEval_EvalFrameDefault + 20988
    frame #32: 0x0000000100aefd09 Python`_PyEval_EvalCodeWithName + 1952
    frame #33: 0x0000000100ae59f6 Python`PyEval_EvalCode + 51
    frame #34: 0x0000000100ae32b6 Python`builtin_exec + 592
    frame #35: 0x0000000100a7d7f1 Python`cfunction_vectorcall_FASTCALL + 169
    frame #36: 0x0000000100a4f333 Python`PyVectorcall_Call + 108
    frame #37: 0x0000000100aecaeb Python`_PyEval_EvalFrameDefault + 28754
    frame #38: 0x0000000100aefd09 Python`_PyEval_EvalCodeWithName + 1952
    frame #39: 0x0000000100a4fa51 Python`_PyFunction_Vectorcall + 227
    frame #40: 0x0000000100aef1e6 Python`call_function + 346
    frame #41: 0x0000000100aec5c9 Python`_PyEval_EvalFrameDefault + 27440
    frame #42: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #43: 0x0000000100aef1e6 Python`call_function + 346
    frame #44: 0x0000000100aec5a9 Python`_PyEval_EvalFrameDefault + 27408
    frame #45: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #46: 0x0000000100aef1e6 Python`call_function + 346
    frame #47: 0x0000000100aec671 Python`_PyEval_EvalFrameDefault + 27608
    frame #48: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #49: 0x0000000100aef1e6 Python`call_function + 346
    frame #50: 0x0000000100aec671 Python`_PyEval_EvalFrameDefault + 27608
    frame #51: 0x0000000100a4f8ec Python`function_code_fastcall + 106
    frame #52: 0x0000000100a50d37 Python`object_vacall + 277
    frame #53: 0x0000000100a50e78 Python`_PyObject_CallMethodIdObjArgs + 217
    frame #54: 0x0000000100b08ef9 Python`PyImport_ImportModuleLevelObject + 1564
    frame #55: 0x0000000100aeac95 Python`_PyEval_EvalFrameDefault + 20988
    frame #56: 0x0000000100aefd09 Python`_PyEval_EvalCodeWithName + 1952
    frame #57: 0x0000000100ae59f6 Python`PyEval_EvalCode + 51
    frame #58: 0x0000000100b08379 Python`exec_code_in_module + 21
    frame #59: 0x0000000100b081bd Python`PyImport_ExecCodeModuleWithPathnames + 252
    frame #60: 0x00000001000bf72c driver`webots_ros2_driver::PythonPlugin::getPyWebotsNodeInstance() + 188
    frame #61: 0x00000001000bf4e5 driver`webots_ros2_driver::PythonPlugin::init(webots_ros2_driver::WebotsNode*, std::__1::unordered_map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > >&) + 325
    frame #62: 0x000000010005d2c8 driver`webots_ros2_driver::WebotsNode::init() + 2616
    frame #63: 0x000000010001ea5d driver`main + 637
    frame #64: 0x00007fff2036af3d libdyld.dylib`start + 1
    frame #65: 0x00007fff2036af3d libdyld.dylib`start + 1
```

It should be related to https://github.com/Fluorescence-Tools/tttrlib/issues/5, but adding the `-undefined dynamic_lookup` doesn't help :/

omichel, 2021-12-21:
Since [macOS is given tier 3 support for ROS 2](https://discourse.ros.org/t/macos-support-in-ros-2-galactic-and-beyond/17891/12), I believe we shouldn't make too much efforts to fix this kind of bugs as we may end up bumping into ROS 2 bugs which are not going to be fixed.

ygoumaz, 2023-02-20:
I am closing this issue, as we switched to UTM for macOS. Feel free to reopen if needed.

closed 2023-02-20 by ygoumaz