cyberbotics/webots_ros2 #365: the problem when run ros2 run webots_ros2_driver driver

henryliuliuliu, 2022-01-03:
I am trying to use webot ros2, when I open a world file built by myself, and run 'ros2 run webots_ros2_driver driver',  it reports 'Robot description is not passed, using default parameters.' and then exits. However, when I use the file webots provides or the epuck example world file provided by webots_ros2, it does report the same thing, but would not exit. Could anyone tell me the reason of that?
is this the reason?
[image]

henryliuliuliu, 2022-01-03:
Below is my world wbt file,
`#VRML_SIM R2021b utf8
WorldInfo {
  basicTimeStep 5
  coordinateSystem "NUE"
}
Viewpoint {
  orientation 0.17518231242501445 -0.906324720909402 -0.3845603433557851 3.9441287635980533
  position 14.14891576139225 19.761547538364443 -11.78430246457972
  follow "agiletaur"
}
TexturedBackground {
}
Robot {
  translation 2.86744 0.094561 -4.71043
  rotation 0 1 0 1.57523
  children [
    Lidar {
      translation 0 0.07 -0.16
      children [
        HokuyoUtm30lx {
          translation 0 0.03 -0.01
          enablePhysics FALSE
        }
      ]
    }
    DEF body Group {
      children [
        Shape {
          appearance Appearance {
            material Material {
              diffuseColor 0.666667 0 0.498039
            }
          }
          geometry Box {
            size 0.152 0.09 0.43
          }
        }
        Transform {
          translation 0.132 0 0
          children [
            Shape {
              appearance Appearance {
                material Material {
                  diffuseColor 0 0.333333 0
                }
              }
            }
          ]
        }
        Transform {
          translation -0.132 0 0
        }
        Transform {
          translation 0.184 0 0
        }
        Transform {
          translation -0.184 0 0
        }
      ]
    }
    DEF fr_6 HingeJoint {
      jointParameters HingeJointParameters {
        position 0.2526927863751345
        anchor -0.096 0 0.17
      }
      device [
        PositionSensor {
          name "pos_6"
        }
[... 683 lines cut ...]
    }
  ]
  name "agiletaur"
  boundingObject USE body
  physics Physics {
    density -1
    mass 5
  }
  controller "<extern>"
}
Ramp30deg {
  translation 0.690343 -0.90602 -4.69
  rotation 0 0 1 -0.261795307179586
}
TexturedBackgroundLight {
}
RectangleArena {
  translation 0 -0.15 0
  floorSize 11 16
  floorTileSize 1.2 0.5
  floorAppearance Parquetry {
    type "dark strip"
  }
  wallHeight 0.5
}`

henryliuliuliu, 2022-01-03:
With some experiments, I found that if I use a parallel leg robot as shown in this picture like the minitaur robot, 
[image]
it does not work. 
But if I use a spot like robot,  then it works:
[image],
then it works

lukicdarkoo, 2022-01-04:
@henryliuliuliu Could you please upload the world file?

henryliuliuliu, 2022-01-06:
Here is the world file
[agiletaur.zip](https://github.com/cyberbotics/webots_ros2/files/7819085/agiletaur.zip)
 

Text:
#VRML_SIM R2021b utf8
WorldInfo {
  basicTimeStep 5
  coordinateSystem "NUE"
}
Viewpoint {
  orientation 0.17518231242501445 -0.906324720909402 -0.3845603433557851 3.9441287635980533
  position 14.14891576139225 19.761547538364443 -11.78430246457972
  follow "agiletaur"
}
TexturedBackground {
}
Robot {
  translation 2.86744 0.094561 -4.71043
  rotation 0 1 0 1.57523
  children [
    Lidar {
      translation 0 0.07 -0.16
      children [
        HokuyoUtm30lx {
          translation 0 0.03 -0.01
          enablePhysics FALSE
        }
      ]
    }
    DEF body Group {
      children [
        Shape {
          appearance Appearance {
            material Material {
              diffuseColor 0.666667 0 0.498039
            }
          }
          geometry Box {
            size 0.152 0.09 0.43
          }
        }
        Transform {
          translation 0.132 0 0
          children [
            Shape {
              appearance Appearance {
                material Material {
                  diffuseColor 0 0.333333 0
                }
              }
            }
          ]
        }
        Transform {
          translation -0.132 0 0
        }
        Transform {
          translation 0.184 0 0
        }
        Transform {
          translation -0.184 0 0
        }
      ]
    }
    DEF fr_6 HingeJoint {
      jointParameters HingeJointParameters {
        position 0.2526927863751345
        anchor -0.096 0 0.17
      }
[... 687 lines cut ...]
    }
  ]
  name "agiletaur"
  boundingObject USE body
  physics Physics {
    density -1
    mass 5
  }
  controller "<extern>"
}
Ramp30deg {
  translation 0.690343 -0.90602 -4.69
  rotation 0 0 1 -0.261795307179586
}
TexturedBackgroundLight {
}
RectangleArena {
  translation 0 -0.15 0
  floorSize 11 16
  floorTileSize 1.2 0.5
  floorAppearance Parquetry {
    type "dark strip"
  }
  wallHeight 0.5
}

henryliuliuliu, 2022-01-06:
is this problem caused by this world file,  then how can I achieve the same mechanism as shown in the first picture, I want two motors on robot body,(the center of yellow horizontal line) to control the two seperate yellow leg), But there is no motor on the hingejoint between yellow leg and red leg.

lukicdarkoo, 2022-01-08:
I have just checked and the problem is related to the URDF exportation which crashes Webots. `SolidReference` nodes are not properly handled in the URDF exporter. We have to fix this issue in the Webots core.

@henryliuliuliu As a temporary solution, it should work by disabling the following line:
https://github.com/cyberbotics/webots_ros2/blob/3a91326c3df2597a1a217d82ab1a60cdd7ce976b/webots_ros2_driver/src/WebotsNode.cpp#L115

closed 2022-01-12 by lukicdarkoo