cyberbotics/webots_ros2 #366: Wrong transform of LiDAR sensor

james-yoo, 2022-01-05:
**Describe the Bug**
LiDAR scan data outputs not correct. I think the reason is that the coordinate system was changed to ENU/FLU in the R2022a version.

**Steps to Reproduce**
1. Install webots R2022a (binary)
2. Source build webots_ros2(branch: master)
3. Follow the turtlebot3 navigation2 instriction [here](https://github.com/cyberbotics/webots_ros2/wiki/Navigate-TurtleBot3)
4. Check LaserScan data on rviz2

**Expected behavior**
LiDAR scan data should be matched with map.

**Affected Packages**
List of affected packages:
  - webots_ros2_turtlebot3
  - webots_ros2_driver
  - ros-foxy-turtlebot3-navigation2

**Screenshots**
[image]

**System**
 - Webots Version: R2022a
 - ROS Version: Foxy Fossa(debian install)
 - Operating System: Linux Ubuntu 20.04

lukicdarkoo, 2022-01-05:
@james-yoo Thank you for reporting the issue! Please let us know whether #367 works for you.

james-yoo, 2022-01-05:
@lukicdarkoo Great! It works now! Thank you.

PR #367 (2022-01-06): Fix LiDAR transform
Fixes #366

**Tasks**
- [x] Fix LiDAR transform
- [x] Add more LiDAR tests
- [x] Verify the related tutorials manually
- [x] Verify point cloud messages
- [x] Revert temporary deleted MoveIt dependency
- Not related problems:
  - [x] Fix Webots version
  - [x] Fix the missing transforms

**Note**
- The tests are passing without the MoveIt dependency, so it seems there is a temporary problem on the ROS side.
- It might be possible that we inverted the order of the LiDAR scan in Webots.
- We should do another release with this fix as soon as possible (before ROS syncs).
closed 2022-01-06 by lukicdarkoo