cyberbotics/webots_ros2 #368: The publication of "/clock" inside "WebotsNode" should be optional.

TaoYibo1866, 2022-01-08:
I find that webots_ros2_driver publishes simulation time to "/clock" automatically, and this is fine when only one robot exists. However, when there are multiple robots present, there will be duplication in simulation time publication. May be it is better to make time publication optional?

lukicdarkoo, 2022-01-08:
This is a good idea, thanks!

BenjaminHug8, 2022-01-10:
I will take care of that with a new feature I am currently working on (it will add a unique supervisor to each simulation to introduce new commands, so I will make it responsible to publish the "/clock" topic).

lukicdarkoo, 2022-01-15:
@TaoYibo1866 As a temporary solution, you can remap the `/clock` topic to some dummy name (e.g. `/dummy/clock`) of all robots except one.

TaoYibo1866, 2022-01-19:
> 

Thank you! That's a good idea.

BenjaminHug8, 2022-02-04:
PR #395 has been merge in #345. Once this last PR will be merged your issue will be solved on the `develop` branch.

BenjaminHug8, 2022-02-16:
@TaoYibo1866 #345 has been merged!

I will close this issue as the changes requested here are present in the `develop` branch, but feel free to reopen if needed.

closed 2022-02-16 by BenjaminHug8