cyberbotics/webots_ros2 #371: Error while building packages

ata2151, 2022-01-12:
**Describe the Bug**
While building the Webots_ros2 packages, on the colcon build step I get the following errors. These are the steps I took:

```
source /opt/ros/$ROS_DISTRO/local_setup.bash

# Retrieve the sources
cd /path/to/ros2_ws
git clone --recurse-submodules https://github.com/cyberbotics/webots_ros2.git src/webots_ros2

# Install dependencies
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO

# Building packages
#ERROR ON THIS STEP
colcon build

# Source this workspace (careful when also sourcing others)
source install/local_setup.bash
```

**Expected behavior**
I expected the colcon build to succeed. 

**Affected Packages**
List of affected packages:
  - webots_ros2_driver
  

**Screenshots**

```
amen@ubuntu:~/webot_ws$ colcon build
Starting >>> webots_ros2_msgs
Starting >>> webots_ros2_importer
Finished <<< webots_ros2_msgs [0.47s]                                      
Starting >>> webots_ros2_driver
Starting >>> webots_ros2_core
Finished <<< webots_ros2_importer [0.74s]                           
--- stderr: webots_ros2_driver                                              
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libController.so when searching for -lController
/usr/bin/ld: cannot find -lController
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppController.so when searching for -lCppController
/usr/bin/ld: cannot find -lCppController
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libdriver.so when searching for -ldriver
/usr/bin/ld: cannot find -ldriver
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppDriver.so when searching for -lCppDriver
/usr/bin/ld: cannot find -lCppDriver
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libcar.so when searching for -lcar
/usr/bin/ld: cannot find -lcar
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppCar.so when searching for -lCppCar
/usr/bin/ld: cannot find -lCppCar
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/webots_ros2_driver_imu.dir/build.make:197: libwebots_ros2_driver_imu.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:248: CMakeFiles/webots_ros2_driver_imu.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libController.so when searching for -lController
/usr/bin/ld: cannot find -lController
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppController.so when searching for -lCppController
/usr/bin/ld: cannot find -lCppController
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libdriver.so when searching for -ldriver
/usr/bin/ld: cannot find -ldriver
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppDriver.so when searching for -lCppDriver
/usr/bin/ld: cannot find -lCppDriver
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libcar.so when searching for -lcar
/usr/bin/ld: cannot find -lcar
/usr/bin/ld: skipping incompatible /home/amen/webot_ws/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppCar.so when searching for -lCppCar
/usr/bin/ld: cannot find -lCppCar
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/driver.dir/build.make:358: driver] Error 1
make[1]: *** [CMakeFiles/Makefile2:221: CMakeFiles/driver.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< webots_ros2_driver [0.69s, exited with code 2]
Aborted  <<< webots_ros2_core [0.79s]                      

Summary: 2 packages finished [1.36s]
  1 package failed: webots_ros2_driver
  1 package aborted: webots_ros2_core
  1 package had stderr output: webots_ros2_driver
  9 packages not processed
```

**System**
 - Webots Version: 1.2.1
 - ROS Version: Galactic
 - Operating System: Ubuntu 20.04.3 arm64
 - Graphics Card:

BenjaminHug8, 2022-01-12:
Hi @ata2151, I can build the packages with the same OS but using [Webots R2022a](https://github.com/cyberbotics/webots/releases/tag/R2022a) which is currently the version target by `webots_ros2`.

Are you really working with Webots 1.2.1 and not an open source version of Webots (the oldest open source version is `R2019a` normally)?

lukicdarkoo, 2022-01-12:
> Operating System: Ubuntu 20.04.3 arm64

Webots is not supported on the ARM platform.

ata2151, 2022-01-12:
> Hi @ata2151, I can build the packages with the same OS but using [Webots R2022a](https://github.com/cyberbotics/webots/releases/tag/R2022a) which is currently the version target by `webots_ros2`. [...]

I am working with Webots R2022a. I mistakenly wrote the version of the webots_ros2 interface. @lukicdarkoo mentioned the Webots is not supported on arm architecture. Are you using the Ubuntu on an arm machine?

BenjaminHug8, 2022-01-12:
> Are you using the Ubuntu on an arm machine?

No sorry, I read a bit too fast and was focus on your Webots version so I miss the fact you were on arm machine.

closed 2022-01-12 by ata2151