cyberbotics/webots_ros2 #372: No reponse through ros2control position controller

henryliuliuliu, 2022-01-13:
**Describe the Bug**
I am trying to use ros2control to control the robot in the webots, I successful control it through velocity and effort control, but when I try to use position controller, the robot does not move. 
I think the ros2control node is started correctly, as shown in the following picture. 
[image]
This is my yaml file for ros2webots:
`
controller_manager:
  ros__parameters:
    update_rate: 100 # Hz

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    joint0_position_controller:
      type: position_controllers/JointGroupPositionController

    joint0_velocity_controller:
      type: velocity_controllers/JointGroupVelocityController

    joint0_effort_controller:
      type: effort_controllers/JointGroupEffortController
    
    joint1_position_controller:
      type: position_controllers/JointGroupPositionController

    joint1_velocity_controller:
      type: velocity_controllers/JointGroupVelocityController

    joint1_effort_controller:
      type: effort_controllers/JointGroupEffortController

    group_position_controller:
      type: position_controllers/JointGroupPositionController

    group_velocity_controller:
      type: velocity_controllers/JointGroupVelocityController

    group_effort_controller:
      type: effort_controllers/JointGroupEffortController

joint0_position_controller:
  ros__parameters:
    joints:
      - motor1

joint0_velocity_controller:
  ros__parameters:
    joints:
      - motor1

joint0_effort_controller:
  ros__parameters:
    joints:
      - motor1

joint1_position_controller:
  ros__parameters:
    joints:
      - motor2

joint1_velocity_controller:
  ros__parameters:
    joints:
      - motor2

joint1_effort_controller:
  ros__parameters:
    joints:
      - motor2

group_position_controller:
  ros__parameters:
    joints:
      - motor1
      - motor2

group_velocity_controller:
  ros__parameters:
    joints:
      - motor1
      - motor2

group_effort_controller:
  ros__parameters:
    joints:
      - motor1
      - motor2
`

Below is my  urdf for my world file.
`
<?xml version="1.0" ?>
<robot name="singleleg">
    <webots>
        

        

        

        

        <plugin type="webots_ros2_control::Ros2Control" />
    </webots>

    <ros2_control name="WebotsControl" type="system">
        <hardware>
            <plugin>webots_ros2_control::Ros2ControlSystem</plugin>
        </hardware>
        <joint name="motor1">
            <state_interface name="position"/>
            <state_interface name="velocity"/>
            <state_interface name="effort"/>

            <command_interface name="position"/>
            <command_interface name="velocity"/>
            <command_interface name="effort"/>
        </joint>
        <joint name="motor2">
            <state_interface name="position"/>
            <state_interface name="velocity"/>
            <state_interface name="effort"/>

            <command_interface name="position"/>
            <command_interface name="velocity"/>
            <command_interface name="effort"/>
        </joint>
    </ros2_control>
</robot>

`
I cannot figure out where is the mistake? Btw, I tried to control it through controller of webots instead of using ros2control, then setPosition function works pretty well. 

Thanks for your time.

BenjaminHug8, 2022-01-14:
> I think the ros2control node is started correctly

Can you show the output of the terminal after the nodes have started when you control with position?

Otherwise your URDF and yaml files look good to me, so I think the issue might come from your ROS controller.

You might check our [universal robot example](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_universal_robot) that is controlled with position to compare with your implementation.
Check the simplest scenario [robot_launch](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/launch/robot_launch.py) and in particular how the robot is controlled [here](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/webots_ros2_universal_robot/ur5e_controller.py).

BenjaminHug8, 2022-03-04:
Hi @henryliuliuliu, is your issue solved?

BenjaminHug8, 2022-03-09:
I will close this issue as the user do not answer anymore.

Feel free to reopen if necessary.

closed 2022-03-09 by BenjaminHug8