cyberbotics/webots_ros2 #374: [ERROR] [driver-2]: process has died [pid 21339, exit code 1, cmd '/opt/ros/foxy/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_lohaksf8 --params-file /tmp/launch_params_688nby5h'].

MichaelOhn, 2022-01-17:
**Describe the Bug**
Hi, firstly thanks for your amazing projects, I've been able to play with webots recently, and I'm really glad about it.
That said, I'm encountering an error when trying to launch webot and ros2 for dji mavic 2 example.

**Steps to Reproduce**
1. Launch `ros2 launch webots_ros2_mavic robot_launch.py`
2. Wait for webots to render the drone.
3. Scroll down to the last message showed in the terminal.
4. See error: `[ERROR] [driver-2]: process has died [pid 21339, exit code 1, cmd '/opt/ros/foxy/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_lohaksf8 --params-file /tmp/launch_params_688nby5h'].`

**Expected behavior**
Be able to pilot the dji mavic 2 drone and consult the available ros2 service list.

**Affected Packages**
List of affected packages:
  - webots_ros2_mavic

**Screenshots**
[image]

[image]

**System**
 - Webots Version: R2019b, 
 - ROS Version: foxy 
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVD7 / Intel® HD Graphics 4000 (IVB GT2)

BenjaminHug8, 2022-01-17:
I see that you have Webots `R2019b` and I suppose you recently clone `webots_ros2`, thus getting the most recent release of `webots_ros2` which is `1.2.1`.

The most recent version of `webots_ros2` targets the newest version of Webots, namely [R2022a](https://github.com/cyberbotics/webots/releases/tag/R2022a).

You have 2 solutions:
- Update your Webots version to `R2022a` (be careful, we changed the axes system orientation to be `Z` up starting from this version, so you might need to adapt code and worlds, see [here](https://github.com/cyberbotics/webots/wiki/How-to-adapt-your-world-or-PROTO-to-Webots-R2022a)).
- Change the version of `webots_ros2` by `git checkout tag`, where `tag` should be a version released during the second half of 2019 (unfortunately the oldest version of `webots_ros2` is [1.0.2](https://github.com/cyberbotics/webots_ros2/releases)from 2020 so I am really not sure that this solution will work, but it might. The command git would be `git checkout 1.0.2` in this case).

MichaelOhn, 2022-01-17:
Hi, 

Thanks for the prompt answer ! Looking for the `webots_ros2` version, I believe I have the version `1.1.3` (installation procedure used [here](https://github.com/cyberbotics/webots_ros2/wiki/Getting-Started)). 
`Webots 2021b` was installed by launching the mavic example automatically (in `/home/user/.ros`). And totally my bad it's not 2019b but 2021b. 
Also, I did have webots R2022a manually installed first, but I had the exact same issue. 
Here next how I found the `webots_ros2` version:
```/opt/ros/foxy/share/webots_ros2 cat package.xml 
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format2.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>webots_ros2</name>
  <version>1.1.3</version>
  <description>Interface between Webots and ROS2</description>

  <maintainer email="support@cyberbotics.com">Cyberbotics</maintainer>
  <license>Apache License 2.0</license>
  <url type="website">http://wiki.ros.org/webots_ros2</url>
  <url type="bugtracker">https://github.com/cyberbotics/webots_ros2/issues</url>
  <url type="repository">https://github.com/cyberbotics/webots_ros2</url>

  <exec_depend>rclpy</exec_depend>
  <exec_depend>std_msgs</exec_depend>
  <exec_depend>builtin_interfaces</exec_depend>
  <exec_depend>webots_ros2_control</exec_depend>
  <exec_depend>webots_ros2_core</exec_depend>
  <exec_depend>webots_ros2_driver</exec_depend>
  <exec_depend>webots_ros2_epuck</exec_depend>
  <exec_depend>webots_ros2_importer</exec_depend>
  <exec_depend>webots_ros2_mavic</exec_depend>
  <exec_depend>webots_ros2_msgs</exec_depend>
  <exec_depend>webots_ros2_tesla</exec_depend>
  <exec_depend>webots_ros2_tiago</exec_depend>
  <exec_depend>webots_ros2_turtlebot</exec_depend>
  <exec_depend>webots_ros2_universal_robot</exec_depend>

  <test_depend>ament_copyright</test_depend>
  <test_depend>ament_flake8</test_depend>
  <test_depend>ament_pep257</test_depend>
  <test_depend>python3-pytest</test_depend>
  <test_depend>webots_ros2_tests</test_depend>

  <export>
    <build_type>ament_python</build_type>
  </export>
</package>```

MichaelOhn, 2022-01-17:
Oh and if it's of any use, I would point out that the launch of `ros2 launch webots_ros2_universal_robot multirobot_launch.py` works rather well:
[image]
[image]
[image]

console output:
```
ros2 launch webots_ros2_universal_robot multirobot_launch.py
[INFO] [launch]: All log files can be found below /home/mike/.ros/log/2022-01-17-12-03-32-728927-dancesurf-37334
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots-1]: process started with pid [37337]
[INFO] [abb_controller-2]: process started with pid [37339]
[INFO] [ur5e_controller-3]: process started with pid [37343]
[INFO] [driver-4]: process started with pid [37345]
[INFO] [driver-5]: process started with pid [37348]
[INFO] [spawner.py-6]: process started with pid [37351]
[INFO] [spawner.py-7]: process started with pid [37356]
[INFO] [spawner.py-8]: process started with pid [37358]
[INFO] [spawner.py-9]: process started with pid [37360]
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 1)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 1)
[spawner.py-6] [INFO] [1642442613.646263102] [spawner_ur_joint_trajectory_controller]: Waiting for /ur5e/controller_manager services
[spawner.py-8] [INFO] [1642442613.784468259] [spawner_abb_joint_trajectory_controller]: Waiting for /abb/controller_manager services
[spawner.py-7] [INFO] [1642442613.890061859] [spawner_ur_joint_state_broadcaster]: Waiting for /ur5e/controller_manager services
[spawner.py-9] [INFO] [1642442613.894660092] [spawner_abb_joint_state_broadcaster]: Waiting for /abb/controller_manager services
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 2)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 2)
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 3)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 3)
[spawner.py-6] [INFO] [1642442615.661830497] [spawner_ur_joint_trajectory_controller]: Waiting for /ur5e/controller_manager services
[spawner.py-8] [INFO] [1642442615.799090007] [spawner_abb_joint_trajectory_controller]: Waiting for /abb/controller_manager services
[spawner.py-7] [INFO] [1642442615.904472096] [spawner_ur_joint_state_broadcaster]: Waiting for /ur5e/controller_manager services
[spawner.py-9] [INFO] [1642442615.906300363] [spawner_abb_joint_state_broadcaster]: Waiting for /abb/controller_manager services
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 4)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 4)
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 5)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 5)
[spawner.py-6] [INFO] [1642442617.673177920] [spawner_ur_joint_trajectory_controller]: Waiting for /ur5e/controller_manager services
[spawner.py-8] [INFO] [1642442617.810641068] [spawner_abb_joint_trajectory_controller]: Waiting for /abb/controller_manager services
[spawner.py-9] [INFO] [1642442617.918008977] [spawner_abb_joint_state_broadcaster]: Waiting for /abb/controller_manager services
[spawner.py-7] [INFO] [1642442617.918105878] [spawner_ur_joint_state_broadcaster]: Waiting for /ur5e/controller_manager services
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 6)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 6)
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 7)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 7)
[ur5e_controller-3] [INFO] [1642442619.278041631] [ur5e.ur5e_controller]: Waiting for action server...
[abb_controller-2] [INFO] [1642442619.290793916] [abb.abb_controller]: Waiting for action server...
[spawner.py-6] [INFO] [1642442619.683557145] [spawner_ur_joint_trajectory_controller]: Waiting for /ur5e/controller_manager services
[spawner.py-8] [INFO] [1642442619.820857495] [spawner_abb_joint_trajectory_controller]: Waiting for /abb/controller_manager services
[spawner.py-9] [INFO] [1642442619.931891741] [spawner_abb_joint_state_broadcaster]: Waiting for /abb/controller_manager services
[spawner.py-7] [INFO] [1642442619.931984746] [spawner_ur_joint_state_broadcaster]: Waiting for /ur5e/controller_manager services
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 8)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 8)
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 9)
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 9)
[spawner.py-6] [INFO] [1642442621.694023184] [spawner_ur_joint_trajectory_controller]: Waiting for /ur5e/controller_manager services
[spawner.py-8] [INFO] [1642442621.830721099] [spawner_abb_joint_trajectory_controller]: Waiting for /abb/controller_manager services
[driver-4] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 10)
[spawner.py-9] [INFO] [1642442621.946290279] [spawner_abb_joint_state_broadcaster]: Waiting for /abb/controller_manager services
[spawner.py-7] [INFO] [1642442621.946290275] [spawner_ur_joint_state_broadcaster]: Waiting for /ur5e/controller_manager services
[driver-5] Cannot open file: /tmp/webots-37337-S419ur/WEBOTS_SERVER (retry count 10)
[... cut ...]
[driver-4] [INFO] [1642442714.576889771] [ur5e.ur_joint_trajectory_controller]: Goal reached, success!
[ur5e_controller-3] [INFO] [1642442714.585536340] [ur5e.ur5e_controller]: Goal succeeded.
[INFO] [ur5e_controller-3]: process has finished cleanly [pid 37343]
[driver-5] [INFO] [1642442714.939225384] [abb.abb_joint_trajectory_controller]: Goal reached, success!
[abb_controller-2] [INFO] [1642442714.942857207] [abb.abb_controller]: Goal succeeded.
[INFO] [abb_controller-2]: process has finished cleanly [pid 37339]

```

BenjaminHug8, 2022-01-18:
> And totally my bad it's not 2019b but 2021b.

Okay then I tested and it effectively seems that the mavic example does not work.
It is working with the newest version `1.2.1` of `webots_ros2` but foxy has not be sync since.

So either you wait for the next sync of foxy or you can build `webots_ros2` from [source](https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#linux)

BenjaminHug8, 2022-01-19:
From your output it seems to me that your created a package and put it inside `webots_ros2` that already contain packages.

Instead the instruction on `webots_ros2` should lead you to have this kind of structure:

```
home/mike/ros_ws/src/
├─ webots_ros2/
│ ├─ webots_ros2/
│ ├─ webots_ros2_driver/
│ ├─webots_ros2_mavic/
│ ├─(and all the other sub-package of webots_ros2)
└─ (your own packages here)
```

Then you should colcon build from `home/mike/ros_ws/`.

Of course you should be able to change the path `home/mike/ros_ws/src/` into `home/WHAT_EVER_FOLDER(S)/ros_ws/src/`.

MichaelOhn, 2022-01-19:
Oh, I'm indeed a newbie to ros2 ! I thank you a lot for clear explanations. Now it works pretty well !

BenjaminHug8, 2022-01-19:
You are welcome!

I close this issue as it seems to be solved but you can reopen it if needed (or open a new one).

closed 2022-01-19 by BenjaminHug8