cyberbotics/webots_ros2 #375: "WebotsNode" is designed to set "robot_description" parameter of "robot_state_publisher" node which is not always useful.

TaoYibo1866, 2022-01-19:
https://github.com/cyberbotics/webots_ros2/blob/99c090403b29e9102ad8f5afc475f6e99d84ae17/webots_ros2_driver/src/WebotsNode.cpp#L115
https://github.com/cyberbotics/webots_ros2/blob/99c090403b29e9102ad8f5afc475f6e99d84ae17/webots_ros2_driver/src/WebotsNode.cpp#L229-L240

It's very kind that webots_ros2_driver try to set the description file once a robot_state_publisher is launched. However, if i wrote my own urdf and want robot_state_publisher to load that, webots_ros2_driver may override it and set the description file to "mRobot->getUrdf()" which is generated from vrml.

The reason i want to load a urdf without webots dependency is that i want my project to be able to switch between simulation and real world. I have an rviz launch file which runs robot_state_publisher, loads my own urdf and starts rviz2. This works in real world scenario. However, in simulation, when i launch webots and the forementioned launch file together, webots_ros_driver override my urdf parameter and cause rviz2 not showing the right model.

I am not sure i understand the codes correctly. Perhaps this is not the cause of my problem. Hope there is document to help us better understanding the behaviour of webots_ros2_driver, thanks.

BenjaminHug8, 2022-01-19:
Hi @TaoYibo1866, there is probably a way to get back information from

https://github.com/cyberbotics/webots_ros2/blob/99c090403b29e9102ad8f5afc475f6e99d84ae17/webots_ros2_driver/src/WebotsNode.cpp#L44

So we can use it in the call of the function `setAnotherNodeParameter()` you mentioned.

That is setup like this in the `webots_ros2_universal_robot` `robot_launch` launch file. 

https://github.com/cyberbotics/webots_ros2/blob/99c090403b29e9102ad8f5afc475f6e99d84ae17/webots_ros2_universal_robot/launch/robot_launch.py#L38
https://github.com/cyberbotics/webots_ros2/blob/99c090403b29e9102ad8f5afc475f6e99d84ae17/webots_ros2_universal_robot/launch/robot_launch.py#L64-L73

Does it sound ok for your case?
If yes I will need you to check that it works in your case when I will have implement this.

TaoYibo1866, 2022-01-19:
Sorry, I am not sure i understand you correctly. Do you mean use `webots_ur5e_description.urdf` to `setAnotherNodeParameter()`? 
I think that is fine. However, i still hope the urdf i manually pass to `robot_state_publisher` will not be overrode.

TaoYibo1866, 2022-01-19:
Thank you for your reply. Currently, i just rename my `robot_state_publisher` to some other name to bypass this issue.

BenjaminHug8, 2022-01-20:
> However, i still hope the urdf i manually pass to `robot_state_publisher` will not be overrode.

How critical is this?
As you might pass your URDF to `webots_ros2_driver` like [here](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/launch/robot_launch.py#L69), would it be ok for you to reuse this information in the `setAnotherNodeParameter()` function or do you prefer to be able to chose wether if the `setAnotherNodeParameter()` function should be called or not?

TaoYibo1866, 2022-01-20:
It is not critical.
I think it is better giving user freedom to choose whether `setAnotherNodeParameter()` function is called or not.

BenjaminHug8, 2022-01-20:
Hum something strange in that story, how does `robot_state_publisher` have an impact on the robot you see in Rviz?

Normally you start Rviz and give it your URDF and beside you start `robot_state_publisher` with the content of your URDF.

Would it be possible to see your launch files in order understand what is happening?

TaoYibo1866, 2022-01-20:
Thank you, i can show you my project.
[https://github.com/TaoYibo1866/diffdrive_explore/tree/dev](url)
My environment is ubuntu20.04, webots2022a, ros2foxy, webots_ros2 1.2.0.

Use `ros2 launch diffdrive_webots localization.launch.py` you can see the issue.

If you rename "robot_state_publisher" to "robot_state_publisher2", rviz will work well.
https://github.com/TaoYibo1866/diffdrive_explore/blob/92f80070b7fce27196175719f2a6f5e25ec8231e/diffdrive_description/launch/load_description.launch.py#L15

TaoYibo1866, 2022-01-20:
In my case, `robot_state_publisher` load urdf and publish `/robot_description` topic, which is subscribed by rviz2.

TaoYibo1866, 2022-01-20:
Oh ,sorry it's 2022a.

BenjaminHug8, 2022-01-20:
Otherwise on my side I have a lot of warnings for each cases and see the same thing in Rviz, but with `robot_state_publisher` instead of `robot_state_publisher2` I also got:
```
[robot_state_publisher-6] Parsing robot urdf xml string.
[robot_state_publisher-6] Error:   No link elements found in urdf file
[robot_state_publisher-6]          at line 179 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/model.cpp
[robot_state_publisher-6] [WARN] [1642688993.121915160] [robot_state_publisher]: Unable to initialize urdf::model from robot description
```
Which is your issue I guess?

TaoYibo1866, 2022-01-20:
You can see the difference in tf tree. In `robot_state_publisher` case, the tf tree shows frames that are generated from PROTO. On the other hand, in `robot_state_publisher2` case, the tf tree shows frames defined in the urdf file.

BenjaminHug8, 2022-01-20:
May I ask you to show images? On my side I have the exact same tf tree.

BenjaminHug8, 2022-01-20:
But I will take a look inside the `WebotsNode` to see what is happening.

BenjaminHug8, 2022-01-20:
Nvm I have not the same tree.

> May I ask you to show images? On my side I have the exact same tf tree.

BenjaminHug8, 2022-01-20:
Ok now I have the global picture, in your project you have:

1. a first URDF file describing the links and joints of your robot.
2. a first launch file using the first URDF file for your `robot_state_publisher`
3. a second URDF file with the Webots tags of your robot.
4. a second launch file for Webots where `webots_ros2_driver` uses the second URDF.
5. a PROTO file of your robot.

The reason why `mRobot->getUrdf()` is not giving a correct URDF is that your PROTO file is not up to date with your first URDF.

The solutions would be either:

1. Giving an URDF that is the sum of your 2 current URDF files `WebotsNode` could reuse this information instead of using `mRobot->getUrdf()` (it works in your case).
2. Have an option to tell `WebotsNode` to setup `robot_state_publisher` with `mRobot->getUrdf()`.

I think to implement option 2) so users are not surprised by `WebotsNode` setting up `robot_state_publisher` in case they forgot to put the corresponding option in `webots_ros2_driver`.

TaoYibo1866, 2022-01-21:
Yes, you are right. I also prefer option (2) because I want to keep the urdf used in rviz without webots tag, in order to keep consistency when switch between simulation and real world.

BenjaminHug8, 2022-01-21:
> I want to keep the urdf used in rviz without webots tag,

Rviz will not parse the Webots tag, you can make one URDF file by mixing your 2 URDF files and give it to `robot_state_publisher` and `webots_ros2_driver` and it won't be a problem.

BenjaminHug8, 2022-01-21:
Hi @TaoYibo1866, this issue should be fixed with #378. I will close this issue but in case something is still not ok for you feel free to reopen!

closed 2022-01-21 by BenjaminHug8