cyberbotics/webots_ros2 #377: Is there any way of modifying dji mavic 2 propellers with ros 2 ?

MichaelOhn, 2022-01-19:
Hi everyone,

**Is your feature request related to a problem? Please describe.**
It is not really an issue, but I would like to explore the dynamics of a drone flight with a ros2 approach and it seems, by listing topics and services, that I cannot directly modify the propeller's angular velocity. Is this available somewhere ?
I would leave this more as a question firstly, since I'm not a ros2 expert and I may have missed the option.

Best regards,

Michael O.

BenjaminHug8, 2022-01-20:
In your case I would create a custom message containing the 4 motor commands so you can create a topic using this message type and send your command. Take a look at this python plugin [here](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_mavic/webots_ros2_mavic/mavic_driver.py) for the mavic. You can remove the control part and instead to subscribe to `/cmd_vel` topic subscribe to your custom topic.

As you said you are not a ros2 expert I would suggest you to take a look at [the official tutorial on msg](https://docs.ros.org/en/foxy/Tutorials/Custom-ROS2-Interfaces.html) and you can also check how we create our custom msg in webots_ros2 [here](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_msgs) and in particular pay attention to the [CMakeLists](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_msgs/CMakeLists.txt) which is the tricky part of creating new msgs in my opinion.

MichaelOhn, 2022-01-20:
Again, thanks a lot Benjamin. I'm going to take a serious look at it !