cyberbotics/webots_ros2 #388: Range-finder returns MAX_RANGE when out of range instead of INFINITY.

TaoYibo1866, 2022-01-28:
[https://github.com/cyberbotics/webots/blob/358d358e83e4d660987307632188832b5c61655a/docs/reference/rangefinder.md?plain=1#L67-L69](url)

I was using Webots R2022a, webots_ros2 1.2.2, and writing a depth camera plugin. The range returned by `getRangeImage()` was MAX_RANGE instead of INFINITY when out of range.

BenjaminHug8, 2022-01-28:
Hi @TaoYibo1866, there is a typo at the line you mentioned. If you take a look [here](https://www.cyberbotics.com/doc/reference/rangefinder?tab-language=c#wb_range_finder_get_range_image), you will see that the range returned is MAX_RANGE when out of range.

I will correct the doc, thanks for spotting this!

TaoYibo1866, 2022-01-28:
[https://github.com/cyberbotics/webots/pull/2509](url)

It seems that Lidar returns INF instead of MAX_RANGE. Does range-finder act differently by design?

TaoYibo1866, 2022-01-28:
> [cyberbotics/webots#2509](https://github.com/cyberbotics/webots/pull/2509)
> 
> It seems that Lidar returns INF instead of MAX_RANGE. Does range-finder act differently by design?

According to Webots R2021a changelog, they should return infinity when the depth is out of range.

BenjaminHug8, 2022-01-28:
Let me check, I do not think Lidar and rangeFinder should have different behavior.

BenjaminHug8, 2022-01-28:
@TaoYibo1866 I investigate a bit. Lidar and rangeFinder should return INF when an object is closer than `minRange` or farther than `maxRange`.

I am currently fixing the rangeFinder and the doc.

BenjaminHug8, 2022-01-28:
This issue should be fixed with https://github.com/cyberbotics/webots/pull/4167.

TaoYibo1866, 2022-01-29:
> This issue should be fixed with [cyberbotics/webots#4167](https://github.com/cyberbotics/webots/pull/4167).

Is there a workaround at hand? Will replace the .so file in webots_ros2/webots_ros2_driver/webots/lib be useful?

BenjaminHug8, 2022-01-31:
Once the fix will be merged, the repos https://github.com/cyberbotics/webots-libcontroller will be updated during the following night.

Then you will be able to navigate to the `webots_ros2/webots_ros2_driver/webots/` folder and git branch on R2022a and git pull.
I will warn you when it will be ready.

BenjaminHug8, 2022-02-04:
@TaoYibo1866 the fix has been merged.

You just have to update your Webots by either:
- Build [Webots yourself](https://github.com/cyberbotics/webots/wiki).
- Or wait for tomorrow. Then you will be able to use the latest [nightly build](https://github.com/cyberbotics/webots/releases) (it will be generated tonight).

BTW forget about the libcontroller, you won't have to change it.

In case you have troubles with the installation feel free to reopen this issue!

closed 2022-02-04 by BenjaminHug8