cyberbotics/webots_ros2 #393: recogniton msg stops updating after a while

jqjiang819, 2022-02-02:
I am using webots R2022a and webots-ros2 on master branch. 
I intended to add a camera with recognition enabled on a turtlebot3_burger PROTO.

The first problem I met is that the camera plugin did not enable recognition before getting objects, and I got the error:
```bash
[driver-5] Error: wb_camera_recognition_get_number_of_objects() called for a disabled device! Please use: wb_camera_recognition_enable()
```

then I changed line 100-101 of `webots_ros2_driver/src/plugins/static/Ros2Camera.cpp` to:
```cpp
    if (recognitionSubscriptionsExist) {
      if (!mIsRecognitionEnabled) {
        printf("enable recognition\n");
        mCamera->recognitionEnable(mPublishTimestepSyncedMs);
        mIsRecognitionEnabled = true;
      }
      publishRecognition();
    } else {
      if (mIsRecognitionEnabled) {
        printf("disable recognition\n");
        mCamera->recognitionDisable();
        mIsRecognitionEnabled = false;
      }
    }
```

it works, but after a short period, it keeps publishing a same message even if I move the robot.
did I misconfigure something?
thanks!

BenjaminHug8, 2022-02-02:
I will take a look at that, thanks for the report !

BenjaminHug8, 2022-02-02:
> The first problem I met is that the camera plugin did not enable recognition before getting objects.

That's true, I will enable it and check that it keeps publishing (might come from your implementation).
I will come back to you once it is solved :)

BenjaminHug8, 2022-02-04:
@jqjiang819 your issue should be fixed with #394. I let you test the `master` branch and in case you still have any issue about the recognition camera please reopen this issue!

closed 2022-02-04 by BenjaminHug8

jqjiang819, 2022-02-04:
The issue is fixed.
Thanks a lot!