cyberbotics/webots_ros2 #399: colcon build failed <<<webots_ros2_driver

iwordly, 2022-02-05:
**Describe the Bug**
A clear and concise description of what the bug is.
n file included from /home/iword/dev_ws/src/webots_ros2/webots_ros2_driver/src/WebotsNode.cpp:25:
/home/iword/dev_ws/src/webots_ros2/webots_ros2_driver/include/webots_ros2_driver/plugins/static/Ros2Camera.hpp:28:10: fatal error: vision_msgs/msg/detection2_d.hpp: No such file or directory
   28 | #include <vision_msgs/msg/detection2_d.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/driver.dir/build.make:76: CMakeFiles/driver.dir/src/WebotsNode.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/driver.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< webots_ros2_driver [14.5s, exited with code 2]

Summary: 7 packages finished [17.0s]
  1 package failed: webots_ros2_driver
  2 packages had stderr output: webots_ros2_driver webots_ros2_importer
  8 packages not processed

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected Packages**
List of affected packages:
  - webots_ros2_driver
  - webots_ros2_importer

**System**
 - ROS Version: foxy 
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce 820M

BenjaminHug8, 2022-02-07:
I took the liberty to edit your message in order to make it more readable.

Otherwise have you followed precisely the instructions to build `webots_ros2` [here](https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#linux)?
It seems that you did not install the dependencies.

BenjaminHug8, 2022-03-04:
Hi @iwordly, is your issue solved?

iwordly, 2022-03-06:
Hi ,I build success now, but it have some error in runing 
`[driver-2] Traceback (most recent call last):
[driver-2]   File "/home/iword/dev_ws/install/my_package/lib/python3.8/site-packages/my_package/my_robot_driver.py", line 30, in step
[driver-2]     rclpy.spin_once(self.__node, timeout_sec=15)
[driver-2] AttributeError: 'MyRobotDriver' object has no attribute '_MyRobotDriver__node'
`
using this command,the robot can't move.
`ros2 topic pub /cmd_vel geometry_msgs/Twist  "linear: { x: 0.1 }"`

BenjaminHug8, 2022-03-07:
> Hi ,I build success now, but it have some error in runing `[driver-2] Traceback (most recent call last): [driver-2] File [...]

May I ask you what are you trying to do, it is hard to guess with only the report of an error in a terminal.

I understand that you are creating your own package, but are you trying to:

- Simply run the `webots_ros2_driver` node?
- Create a python plugin to extend the interface (I guess it is this one, but please confirm)?
- Anything else?

iwordly, 2022-03-07:
thinks you ,I want to run a example in _ros2 documentation:foxy_ this website at follow:
[](https://docs.ros.org/en/foxy/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html)

BenjaminHug8, 2022-03-07:
Ok then it seems you miss a step in the tutorial.

There are 4 steps to do:
1. Modify the python plugin
2. Create the URDF file
3. Create the launch file
4. Modify the setup.py file

For each part, you can simply copy past the code found at the beginning of each section in the file and it should work.
I guess you are missing some lines in a file, check that:
1. The python plugin has the following lines:
```
rclpy.init(args=None)
self.__node = rclpy.create_node('my_robot_driver')
self.__node.create_subscription(Twist, 'cmd_vel', self.__cmd_vel_callback, 1)
```
3. Your URDF file is corresponding to:
```
<?xml version="1.0" ?>
<robot name="My robot">
    <webots>
        <plugin type="my_package.my_robot_driver.MyRobotDriver" />
    </webots>
</robot>
```
Check the underscores or the letter cases, which are important.
4. You have an entry point in the setup.py file for the python plugin:
```
entry_points={
        'console_scripts': [
            'my_robot_driver = my_package.my_robot_driver:main',
        ],
    },
```

Re-check everything closely and tell me if it works.

iwordly, 2022-03-08:
Hi, I am checked every step ,it as same as you say . And I found that in the python plugin file not main() function ,is it this reason?

BenjaminHug8, 2022-03-08:
> And I found that in the python plugin file not main() function ,is it this reason?

Yes it is normal, your plugin will be run by our interface.

> follow your remaind ,I am renew a pkg it also this error: `[driver-2] rclpy.spin_once(self.__node, timeout_sec=15) [driver-2] AttributeError: 'MyRobotDriver' object has no attribute '_MyRobotDriver__node`

Would it be possible for you to zip your package folder and send it to me so I can check it?
It is a bit hard to say without the code.

BenjaminHug8, 2022-03-08:
@iwordly I do not know if your last message is a missclick, but there is no files with it.

iwordly, 2022-03-08:
I am sorry ,I used **qq mail** return to you just now ,I confirm that send zip file. and are you other method send file?

iwordly, 2022-03-08:
[image]

BenjaminHug8, 2022-03-08:
I cannot receive your files when you respond to the mail of Github,  as it simply write a message in your Github issue.

Can you send a mail with your files to support@cyberbotics.com?

BenjaminHug8, 2022-03-08:
First of all I see that your world is in Webots `R2021b`, I encourage you to update to `R2022a` if possible.

Then your python plugin `my_robot_driver.py` should be:
```
import rclpy
from geometry_msgs.msg import Twist

HALF_DISTANCE_BETWEEN_WHEELS = 0.045
WHEEL_RADIUS = 0.025

class MyRobotDriver:
    def init(self, webots_node, properties):
        self.__robot = webots_node.robot

        self.__left_motor = self.__robot.getDevice('left wheel motor')
        self.__right_motor = self.__robot.getDevice('right wheel motor')

        self.__left_motor.setPosition(float('inf'))
        self.__left_motor.setVelocity(0)

        self.__right_motor.setPosition(float('inf'))
        self.__right_motor.setVelocity(0)

        self.__target_twist = Twist()

        rclpy.init(args=None)
        self.__node = rclpy.create_node('my_robot_driver')
        self.__node.create_subscription(Twist, 'cmd_vel', self.__cmd_vel_callback, 1)

    def __cmd_vel_callback(self, twist):
        self.__target_twist = twist

    def step(self):
        rclpy.spin_once(self.__node, timeout_sec=0)

        forward_speed = self.__target_twist.linear.x
        angular_speed = self.__target_twist.angular.z

        command_motor_left = (forward_speed - angular_speed * HALF_DISTANCE_BETWEEN_WHEELS) / WHEEL_RADIUS
        command_motor_right = (forward_speed + angular_speed * HALF_DISTANCE_BETWEEN_WHEELS) / WHEEL_RADIUS

        self.__left_motor.setVelocity(command_motor_left)
        self.__right_motor.setVelocity(command_motor_right)
```

You do not have to add:
```
def main(args=None):
    rclpy.init()
    my_webots_driver = MyWebotsDriver
    rclpy.spin(my_webots_driver)
    my_webots_driver.destroy()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```
in this file.

Otherwise I do not see no other issues, I can build your package and run it.

Let me know if that works for you.

BenjaminHug8, 2022-03-08:
Concerning your other mail, how can you get an error `rclpy.spin_once(self.__node, timeout_sec=15)` when the corresponding line in the plugin stats a timeout of `0`?

Are you sure that you are using the correct files?
(maybe try to remove `/build`,`/log` and `/install` from your ros working space)

RegularTriangle, 2022-03-25:
> I took the liberty to edit your message in order to make it more readable.
> 
> Otherwise have you followed precisely the instructions to build `webots_ros2` [here](https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#linux)? It seems that you did not install the dependencies.

Hi, I meet a similar issue when I try to use the `develop` branch in my workspace(I want to get the global (world) position/orientation of a node with `Supervisor node` in ros-foxy).  Here are my steps:
1. Download the source code(`develop` branch).
2. Extract to my workspace
3. colcon build

Error messages:
[image]

I have install another version by using a package manager.

lukicdarkoo, 2022-03-26:
@RegularTriangle Did you follow instructions from here:
https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#linux

RegularTriangle, 2022-03-27:
> @RegularTriangle Did you follow instructions from here: https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#linux

是的，我遵循了除了获取源代码之外的所有指令
I followed  every instruction except  this: `git clone --recurse-submodules https://github.com/cyberbotics/webots_ros2.git src/webots_ros2`. I downloaded the source code directly from the web page.
[image]

lukicdarkoo, 2022-03-27:
The command you missed also clones git submodules, and these submodules are not available when the downloading zip. I strongly suggest that you try to git clone as it will automatically pull proper git submodule versions. Otherwise, make sure you have `webots_ros2_importer/urdf2webots/urdf2webots` and `webots_ros2_driver/webots` folders downloaded as well.

RegularTriangle, 2022-04-02:
> 

I  tried to git clone with the following command: 
`git clone --recurse-submodules -b develop https://github.com/cyberbotics/webots_ros2.git src/webots_ros2`
And built the packages successfully. 
[image]

However, none of the examples could work successfully. (Wait for `/controller_manager services`  timeout.)
[image]
[image]

closed 2022-07-27 by omichel