cyberbotics/webots_ros2 #401: webots_ros2_turtlebot TF-Tree is split

janx8r, 2022-02-14:
**Describe the Bug**
After launching robot_launch.py in webots_ros2_turtlebot the transformation tree is split.
This leads to the fact that the navigation does not work, because the sensor data of the lidar cannot be transformed to the odometry.
Apparently, the transformations between the frames 'base_link' and 'LDS-01' are missing, among others:
[image]
If I add the missing transforms with static_transform_publisher in robot_launch.py, everything works.
I have the same problem with webots_ros2_tiago.

**Steps to Reproduce**
1. install Webots R-2022a (I have tried the debian and the tarball installation)
2. webots_ros2 clone and build:
    ````mkdir test_workspace
    cd test_workspace
    mkdir src
    cd src
    git clone --recursive https://github.com/cyberbotics/webots_ros2.git
    cd ..
    colcon build
    source install/local_setup.bash````
3. launch webots_ros2_turtlebot:
    ````ros2 launch webots_ros2_turtlebot robot_launch.py````
4. show the TF-Tree on a new terminal:
    ````ros2 run rqt_tf_tree rqt_tf_tree --force-discover````

**Expected behavior**
The tree should be connected in one piece so that the transformations between all frames can be calculated.
It should look like this in full (from @BenjaminHug8  in the Discord channel):
[frames.pdf](https://github.com/cyberbotics/webots_ros2/files/8062497/frames.pdf)

**Affected Packages**
List of affected packages:
  - webots_ros2_turtlebot
  - webots_ros2_tiago
  - probably more robot packages that are not yet tested

**System**
 - Webots Version: R-2022a
 - ROS Version: galactic
 - Operating System: Ubuntu 20.04
 - Graphics Card: NVIDIA Corporation GF108GLM [NVS 5200M]

**Additional context**
My outputs from webots_ros2_turtlebot:
````
jan@jan-laptop:~/ros2_test_ws$ ros2 launch webots_ros2_turtlebot robot_launch.py
[INFO] [launch]: All log files can be found below /home/jan/.ros/log/2022-02-14-17-27-01-907517-jan-laptop-86755
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [spawner.py-1]: process started with pid [86757]
[INFO] [spawner.py-2]: process started with pid [86759]
[INFO] [webots-3]: process started with pid [86761]
[INFO] [robot_state_publisher-4]: process started with pid [86763]
[INFO] [driver-5]: process started with pid [86765]
[INFO] [static_transform_publisher-6]: process started with pid [86769]
[driver-5] Cannot open file: /tmp/webots-86761-HNGQX4/WEBOTS_SERVER (retry count 1)
[static_transform_publisher-6] [INFO] [1644856022.536536500] [static_transform_publisher_uw8YMyY3at2XRYQe]: Spinning until killed publishing transform from 'base_link' to 'base_footprint'
[robot_state_publisher-4] [INFO] [1644856022.541359329] [robot_state_publisher]: got segment 
[spawner.py-2] [INFO] [1644856022.821224693] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1644856022.840007177] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[driver-5] Cannot open file: /tmp/webots-86761-HNGQX4/WEBOTS_SERVER (retry count 2)
[driver-5] Cannot open file: /tmp/webots-86761-HNGQX4/WEBOTS_SERVER (retry count 3)
[spawner.py-2] [INFO] [1644856024.836095716] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1644856024.855167190] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[driver-5] Cannot open file: /tmp/webots-86761-HNGQX4/WEBOTS_SERVER (retry count 4)
[driver-5] Cannot open file: /tmp/webots-86761-HNGQX4/WEBOTS_SERVER (retry count 5)
[spawner.py-2] [INFO] [1644856026.847329130] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1644856026.866124079] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[driver-5] Cannot open file: /tmp/webots-86761-HNGQX4/WEBOTS_SERVER (retry count 6)
[driver-5] Cannot open file: /tmp/webots-86761-HNGQX4/WEBOTS_SERVER (retry count 7)
[spawner.py-2] [INFO] [1644856028.859561341] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1644856028.878329833] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[... 80 lines cut ...]
/home/jan/ros2_test_ws/install/webots_ros2_mavic/lib/python3.8/site-packages:
/home/jan/ros2_test_ws/install/webots_ros2_epuck/lib/python3.8/site-packages:
/home/jan/ros2_test_ws/install/webots_ros2_driver/lib/python3.8/site-packages:
/home/jan/ros2_test_ws/install/webots_ros2_core/lib/python3.8/site-packages:
/home/jan/ros2_test_ws/install/webots_ros2_msgs/lib/python3.8/site-packages:
/home/jan/ros2_test_ws/install/webots_ros2_importer/lib/python3.8/site-packages:
/opt/ros/galactic/lib/python3.8/site-packages:
/usr/local/webots/lib/controller/python38:
````

`echo $WEBOTS_HOME`: 
````/usr/local/webots````

`echo $LD_LIBRARY_PATH`: 
````
/home/jan/ros2_test_ws/install/webots_ros2_control/lib:
/home/jan/ros2_test_ws/install/webots_ros2_driver/lib:
/home/jan/ros2_test_ws/install/webots_ros2_msgs/lib:
/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:
/opt/ros/galactic/opt/yaml_cpp_vendor/lib:
/opt/ros/galactic/opt/rviz_ogre_vendor/lib:
/opt/ros/galactic/lib/x86_64-linux-gnu:
/opt/ros/galactic/lib:
/usr/local/webots/lib/controller:
````

BenjaminHug8, 2022-02-15:
Hi @janx8r, thanks for the report, I will investigate that!

BenjaminHug8, 2022-02-15:
Thanks for that detailed report, for now I still get the same result as you posted (the pdf file).

My outputs from webots_ros2_turtlebot is the same than yours but that is weird than you do not have a more complete tree (robot_state_publisher is noticing more parts of the robots).

One thing I noticed is that you might miss some dependencies. In the [instructions](https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#linux) there is 2 additional commands:
```
# Install dependencies
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO
```

May I ask you to do it? I am not sure that the problem is here but I want to be sure.

janx8r, 2022-02-15:
Thanks for your help!
Unfortunately the updates were not the problem, everything was there.
Is it actually correct that the submodule `webots_ros2/webots_ros2_driver/webots` points to `8b3d028` and not to `R2022a`?

BenjaminHug8, 2022-02-15:
> Is it actually correct that the submodule `webots_ros2/webots_ros2_driver/webots` points to `8b3d028` and not to `R2022a`?

Submodules always target a commit rather than a branch (`8b3d028` is a commit and `R2022a` a branch). It could be interesting for you to try to checkout on `R2022a` and `git pull`, but doing so will target the `master` branch version of Webots so you might want to try to build Webots from [sources](https://github.com/cyberbotics/webots/wiki/Linux-installation/).

janx8r, 2022-02-15:
I just built webots from the sources (master branch) and checked out the driver on R2022a. Unfortunately I still get the same disconnected tree.
So far I have not been able to understand how the TF tree is generated. I assume that the information from the PROTO-file is converted into transformations. In which package/software module does this happen? Maybe I can debug it then.

BenjaminHug8, 2022-02-16:
The idea is that you can ask the `webots_ros2_driver` node to set the `robot_description` of the `robot_state_publisher` node like here for Turtlebot:
https://github.com/cyberbotics/webots_ros2/blob/ebf6c40e890eb295a8cb15aac3b79b4f61e5f472/webots_ros2_turtlebot/launch/robot_launch.py#L69

The driver node will check this flag here:
https://github.com/cyberbotics/webots_ros2/blob/ebf6c40e890eb295a8cb15aac3b79b4f61e5f472/webots_ros2_driver/src/WebotsNode.cpp#L45

And take the urdf generated by Webots here:
https://github.com/cyberbotics/webots_ros2/blob/ebf6c40e890eb295a8cb15aac3b79b4f61e5f472/webots_ros2_driver/src/WebotsNode.cpp#L116-L117

BenjaminHug8, 2022-02-16:
I just noticed you were using `galactic` and not `foxy`, I can reproduce the bug with `galactic`. I will check but probably there is a change on how the `robot_state_publisher` works now.

BenjaminHug8, 2022-02-16:
Ok the issue will be fixed here https://github.com/ros/robot_state_publisher/pull/189.

You might clone this branch meanwhile to make it work!

janx8r, 2022-02-16:
Great, that solves the problem for me too!
Here are the steps I have done:
````
cd src
git clone https://github.com/ros/robot_state_publisher.git
cd robot_state_publisher
sudo apt install git-extras
git pr 189
cd ../..
colcon build
````
This should only be necessary until the PullRequest #189 is merged and the update is available via apt.
Many thanks Benjamin for your help!

Should I close the issue now or wait until #189 is merged? If I leave it open, it may help other users.

BenjaminHug8, 2022-02-16:
> This should only be necessary until the PullRequest #189 is merged and the update is available via apt. Many thanks Benjamin for your help!

You are welcome @janx8r!

> Should I close the issue now or wait until #189 is merged? If I leave it open, it may help other users.

People can search in the closed issues and they are also pretty well referenced on Google, so I would rather close this issue as a solution has been found (it might sounds this issue is not solved if we keep it opened).

closed 2022-02-16 by janx8r