cyberbotics/webots_ros2 #403: Future of PROTO robot descriptions unclear

ruffson, 2022-02-22:
**Documentation page(s) concerned**
All pages.

**What is the problem with the current documentation?**
I am confused about the current use of PROTO files for robot descriptions when using Webots 2022a and ros2 galactic.

All examples and tutorial code in this repo relies on URDF files. Even the foxy branch demos which included [PROTO files](https://github.com/cyberbotics/webots_ros2/tree/foxy/webots_ros2_epuck/protos) for epuck still relied on an URDF representation for the `robot_description`.

There is no documented way of generating a `robot_description` from PROTO files. In fact, there is little mention of PROTO files in the documentation or the driver code for `webots_ros2` in general. The only mention in the wiki is [how to convert a urdf to proto](https://github.com/cyberbotics/webots_ros2/wiki/Using-URDF-or-Xacro). It seems a wiki entry with a migration guide from PROTO to URDF would seem more helpful.

It is not clear if PROTO's have any future for workflows with ros2. But it certainly makes little sense to keep the same robot representation in both formats. Last year, it was discussed [here](https://discourse.ros.org/t/urdf-with-webots/19665) how to deal with PROTO and URDF files in webots going forward, but I could not find any information on the webots website about any change in direction about the topic.

While I appreciate one common format, it would be helpful to clear up the following questions:
- What are roughly the steps to migrate from a ROS1/Webots2021 using PROTO files to ROS2/Webots2022?
- Is it possible to use one or the other format or is a URDF always required?
- What is the best way to convert a complex PROTO that includes other PROTOs of devices e.g. LiDARs, etc. to URDF? Exporting from webots does not seem to export the whole geometry or semantics of the devices, just the names, origins and primitive geometries. (e.g. how would one deal with the polygons defined in an IndexedFaceSet?)

Again, I welcome the effort towards a truly universal format, but it is hard to make a transition with so many changes and unknowns.

BenjaminHug8, 2022-02-23:
> All examples and tutorial code in this repo relies on URDF files. Even the foxy branch demos which included [PROTO files](https://github.com/cyberbotics/webots_ros2/tree/foxy/webots_ros2_epuck/protos) for epuck still relied on an URDF representation for the `robot_description`.

The foxy, rolling and galactic branches are no longer used, everything is based on the `master` branch. But that's right we should make it clear.

The examples relies on URDF files for mostly for plugins ( `ros2control`, `webots_ros2_driver` configurations etc...), but they do not contain the model of the robot as it is contained in the PROTO files shipped with Webots (for example the `Tesla` example uses a Tesla PROTO in the simulation but has a URDF file for a python plugin to extend the ROS 2 interface).

> There is no documented way of generating a `robot_description` from PROTO files. In fact, there is little mention of PROTO files in the documentation or the driver code for `webots_ros2` in general. [...]

In fact, you can go from URDF to PROTO with the tool you mentioned, but the other way around is not directly accessible. What I mean is that a `robot_description` parameter is created by the Webots API function `getUrdf()` and can be sent to the `robot_state_publisher` node here (this is done with an option of the `webots_ros2_driver` ):
https://github.com/cyberbotics/webots_ros2/blob/ebf6c40e890eb295a8cb15aac3b79b4f61e5f472/webots_ros2_driver/src/WebotsNode.cpp#L117

You should be able to use this `getUrdf()` function in a python or C++ plugin but I do not why you would need that except for the `robot_state_publisher` node, but please tell me if you do have a use case.

Does a guide explaining a little bit more what are the options for the `webots_ros2_driver` node would have helped you (in particular the option setting the `robot_description` parameter)?

> It is not clear if PROTO's have any future for workflows with ros2. But it certainly makes little sense to keep the same robot representation in both formats. Last year, it was discussed [...]

PROTO format is used in Webots to easily reuse robots, but basically that is an extension of the VRML format, format used by Webots for the world files. Thus PROTOs will stay in Webots whatever happens in webots_ros2.

That's right that keeping 2 formats with the same information is not handy, as approved by all the users in the discussion you mentioned. The idea of relying only on the URDF file has been implemented this [PR](https://github.com/cyberbotics/webots_ros2/pull/345) that has been merged in `develop`. It should be merged in `master` and released with the next release (R2022b) of Webots this summer. The global idea is that a new `Ros2Supervisor` node has been added that makes the bridge between Webots and ROS 2, and then you can request this new node to spawn a robot modeled in a URDF file.

> While I appreciate one common format, it would be helpful to clear up the following questions:
> 
>     - What are roughly the steps to migrate from a ROS1/Webots2021 using PROTO files to ROS2/Webots2022?

I will split that in two parts:

- **Transition from Webots 2021 to Webots 2022:** The big change here is that Webots 2022 is in FLU, that means before the axis system was `Y` up and now it is `Z` up. This has been decided in order to have the same axis system than ROS and many other tools of today. You can check this guide [here](https://github.com/cyberbotics/webots/wiki/How-to-adapt-your-world-or-PROTO-to-Webots-R2022a) to make the transition.
- **Transition from ROS 1 to ROS 2:** This part is inherent to ROS and they have documentation/tutorial for it, so I guess it is better to link it to you [here](https://docs.ros.org/en/foxy/Contributing/Migration-Guide.html).

>     - Is it possible to use one or the other format or is a URDF always required?

With the new feature to import URDF robot, you will be able to only use URDF. But for now you have to rely on the PROTO file too as it is needed by Webots to include your robot in the simulation, and you may want to rely on a URDF file in case you use other ROS 2 nodes that require it or if you want to parametrize the webots_ros2_driver interface, like [here](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Creating-a-Custom-Package).

> - What is the best way to convert a complex PROTO that includes other PROTOs of devices e.g. LiDARs, etc. to URDF? Exporting from webots does not seem to export the whole geometry or semantics of [...]

As shown in the [description](https://cyberbotics.com/doc/reference/robot#wb_robot_get_urdf) of the function to get the URDF from Webots, it is not complete and thus the whole geometry is not exported. I do not think that including PROTOs in a PROTO is an issue, but using `IndexedFaceSet` or `Mesh` would require some changes to support them.

So to have a complete URDF generated from a PROTO would require to improve the export. Otherwise a way to go is rather to have a URDF file to model the robot and get a PROTO file from it (some PROTO robots of Webots have been created like this).

> Again, I welcome the effort towards a truly universal format, but it is hard to make a transition with so many changes and unknowns.

What transition are you talking about here?
Webots 2021 to Webots 2022, ROS 1 to ROS 2, URDF to PROTO or PROTO to URDF?

ruffson, 2022-02-23:
Thank you for your thorough response!

> Does a guide explaining a little bit more what are the options for the webots_ros2_driver node would have helped you (in particular the option setting the robot_description parameter)?

Absolutely.

> The idea of relying only on the URDF file has been implemented this https://github.com/cyberbotics/webots_ros2/pull/345 that has been merged in develop. It should be merged in master and released with the next release (R2022b) of Webots this summer. The global idea is that a new Ros2Supervisor node has been added that makes the bridge between Webots and ROS 2, and then you can request this new node to spawn a robot modeled in a URDF file.

That is great to hear!

> What transition are you talking about here?
Webots 2021 to Webots 2022, ROS 1 to ROS 2, URDF to PROTO or PROTO to URDF?

Basically ROS1 to ROS2 but since the changes in Webots 2022 (namely the change of frames) are quite valuable, also Webots 2021 to 2022.

The wiki tutorials show [how to create a robot](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Create-Webots-Robot). But the other tutorials which are simulating robots are using the robots that webots already comes with. Confusingly they still rely on an URDF suggesting that they might be self-contained. And since a lot of simulating robots is implicit (e.g. the name of a proto file is used instead of the path or full `my_proto.proto`) it can be tricky to find out what efforts need to be done to make a robot from scratch work with webots.

I think it would be nice to have an intermediate tutorial that shows how a robot which was created in the above tutorial be used in a ros2 project. In the most basic way. Just by launching it and having it publish to some basic topics.

But your explanations give a good idea of the current state of things regarding proto and urdf. 

For me the tl;dr is this:
> With the new feature to import URDF robot, you will be able to only use URDF. But for now you have to rely on the PROTO file too as it is needed by Webots to include your robot in the simulation, [...]

Thanks a lot!

closed 2022-02-23 by ruffson

BenjaminHug8, 2022-02-24:
> > Does a guide explaining a little bit more what are the options for the webots_ros2_driver node would have helped you (in particular the option setting the robot_description parameter)?
> 
> Absolutely.

Noted!

> Basically ROS1 to ROS2 but since the changes in Webots 2022 (namely the change of frames) are quite valuable, also Webots 2021 to 2022.

I suggest to split the transition, first ROS1 to ROS2 using Webots 2021b. Take a look at the official guide ROS1 to ROS2 I cited and compare your implementation with some examples on this GitHub. Maybe start from something very simple in ROS2 like the `Mavic` example and then try to re implement your ROS 1 functionalities step by step. You might also take a look at the [e-puck](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-E-puck-for-ROS2-Beginners) example to understand how to setup things, as the tutorial is very guided.

Once it works you can make the transition between Webots 2021 and 2022 with the link I mentioned .

> I think it would be nice to have an intermediate tutorial that shows how a robot which was created in the above tutorial be used in a ros2 project. In the most basic way. Just by launching it and having it publish to some basic topics.

We have made a tutorial on the official ROS 2 doc [here](https://docs.ros.org/en/foxy/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html). I will add the link in the tutorial to create a robot. This is probably what you are asking :)

ruffson, 2022-02-24:
@BenjaminHug8 Thank you for the time to clear up these issues and give me some helpful pointers on how to proceed! I really appreciate that.