cyberbotics/webots_ros2 #405: Timestep wrongly overwritten

PhilippPolterauer, 2022-03-02:
https://github.com/cyberbotics/webots_ros2/blob/ebf6c40e890eb295a8cb15aac3b79b4f61e5f472/webots_ros2_core/webots_ros2_core/devices/lidar_device.py#L50-L62

The timestep is overwritten with a hardcoded number. I suggest changeing to
`self._timestep = self._get_param('timestep', 1e-2)`

BenjaminHug8, 2022-03-04:
I close this issue as it simply requires to deprecate `webots_ros2_core`, which is already mentioned in https://github.com/cyberbotics/webots_ros2/issues/359

closed 2022-03-04 by BenjaminHug8

PhilippPolterauer, 2022-03-05:
Thanks for your response! We are already transitioning to webots_ros2_driver. It would indeed be helpful to have the preferred approach a bit more highlighted!