cyberbotics/webots_ros2 #406: Make devicemanager extendable

PhilippPolterauer, 2022-03-02:
https://github.com/cyberbotics/webots_ros2/blob/ebf6c40e890eb295a8cb15aac3b79b4f61e5f472/webots_ros2_core/webots_ros2_core/devices/device_manager.py#L47-L61

The current implementation of the DeviceManager only supports a hardcoded list of Devices. 
It would be nice to have a register_device function which would allow adding custom devices with special ros2 handling.

BenjaminHug8, 2022-03-02:
In `webots_ros2_driver` you have the possibility to extend the interface with C++ and Python plugins, you can take a look [here](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Creating-a-Custom-Plugin).

BenjaminHug8, 2022-03-04:
I close this issue as it simply requires to deprecate `webots_ros2_core`, which is already mentioned in https://github.com/cyberbotics/webots_ros2/issues/359

closed 2022-03-04 by BenjaminHug8