cyberbotics/webots_ros2 #407: Does webots_ros2 support SliderJoint?

RegularTriangle, 2022-03-02:
When I use SliderJoint in the Robot node, the ROS2 package, which used to work properly, will report an error.  I want to know how to solve this problem?
[image]
[image]

RegularTriangle, 2022-03-03:
> 
Thank you for your reply, but It seems that changing the DEF for the Transform node can't work. I made the following changes:
[image]
And it reports the the same  error as follow:
[image]

However, when I move the SliderJoint node from the children of the Transform node to the children of the Robot node , it works properly.
[image]
[image]

BenjaminHug8, 2022-03-04:
If you check the error message, it is talking about a `LIFT_TRANSFORM` missing link and not anymore about a `LIFT` missing link, so the error type is the same but overall the error is not the same.

But it seems you manage to have a solution to your issue.

However it seems that the `exportUrdf()` function used by Webots does not manage the scenario using the `sliderJoint` in a `Transform` node.

RegularTriangle, 2022-03-04:
> If you check the error message, it is talking about a `LIFT_TRANSFORM` missing link and not anymore about a `LIFT` missing link, so the error type is the same but overall the error is not the same. [...]

Thanks. I have checked the error message and I think it is the same error. I found that the name of the missing node would change to the same as the DEF of the `Transform` node.
[image]

In a word, a solution to this issue is to avoid using the `SliderJoint` in a `Transform` node.

BenjaminHug8, 2022-03-04:
Perfect, then your issue is solved right (you managed to use the `SliderJoint`)?

(In case your issue is solved, please do not close this issue. We will keep it to track and fix the use of the `Transform` node.)

RegularTriangle, 2022-03-04:
`Perfect, then your issue is solved right (you managed to use the SliderJoint)?`
Yes.
`(In case your issue is solved, please do not close this issue. We will keep it to track and fix the use of the Transform node.)`
OK.

gStewartAtUpower, 2022-07-01:
I think that this bug also applies to HingeJoint objects which are children of Transform objects.

`[robot_state_publisher-3] Error:   Failed to build tree: parent link [slot] of joint [slot_redacted N_joint] not found.  This is not valid according to the URDF spec. Every link you refer to from a joint needs to be explicitly defined in the robot description. To fix this problem you can either remove this joint [slot_redacted N_joint] from your urdf file, or add "<link name="slot" />" to your urdf file.
[robot_state_publisher-3]          at line 225 in /tmp/binarydeb/ros-galactic-urdfdom-2.3.5/urdf_parser/src/model.cpp
[robot_state_publisher-3] Failed to parse robot description using: urdf_xml_parser/URDFXMLParser
[robot_state_publisher-3] terminate called after throwing an instance of 'std::runtime_error'
[robot_state_publisher-3]   what():  Unable to initialize urdf::model from robot description`

----

`                      Transform {
                        translation 0 0 0.2
                        children [
                          HingeJoint {
                            jointParameters HingeJointParameters {
                            }
                            device [
                              RotationalMotor {
                                name "redacted M"
                              }
                            ]
                            endPoint Slot {
                              type "vehicle wheel"
                              endPoint TractorFrontWheel {
                                name "redacted N"
                                tireRadius 0.2
                                width 1.1
                                physics NULL
                              }
                            }`
                          
Can @RegularTriangle help clarify and establish the last known-good version before this bug began to occur? (regarding the "which used to work properly" comment)

ygoumaz, 2023-02-20:
This issue was fully fixed in https://github.com/cyberbotics/webots/pull/5899.

closed 2023-02-20 by ygoumaz