cyberbotics/webots_ros2 #408: webots_ros2_driver and sumo traffic simulation

sebastianohl, 2022-03-02:
**Describe the Bug**
When compiling v1.2.2 (or master) it is possible to control a vehicle (e.g. the tesla example). However, if you would like to add simulated traffic using the sumo simulation (as it is packaged by webots 2022a), there is library conflict. You can only do one thing, control the vehicle or start traffic simulation.

The reason for this library mismatch lies in different source code that is located in webots_ros2_driver/webots (compared between the git repo and the code required for webots 2022a). If the source files are replaced by hand from the files delivered in the webots 2022a binary package, it works well.

**Steps to Reproduce**
1. checkout repo
2. build ros2 modules for ros2 galactic
3. modify tesla example to contain a sumo simulation
4. start simulation
5. starting sumo will fail with a missing symbol
5a. this error can be overcome by setting LD_LIBRARY_PATH to $WEBOTS_HOME/lib/controller. However, this results in not beeing able to run the tesla driver node anymore.

**Expected behavior**
sumo simulation supervisor and webots_ros2_driver control node should run simultaniously (one conroling the robot and the other the traffic). 

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**System**
 - Webots Version: R2022a
 - ROS Version: galactiv
 - Operating System: Ubuntu 20.04
 - Graphics Card: none

BenjaminHug8, 2022-03-08:
Thank you for raising this issue, we will investigate that!

ygoumaz, 2023-03-16:
If I understant correctly, this issue was fixed in #664? I am closing this issue for now, please feel free to reopen it if it is not fixed.

closed 2023-03-16 by ygoumaz