cyberbotics/webots_ros2 #411: Importer introduces mismatch visual and collision models

cesar-lopez-mar, 2022-03-04:
**Describe the Bug**
The collision model of the wheels in a robot model is now miss aligned after conversion to proto. I visualize my urdf in rviz and they are aligned.  This behavior is seen now in version 1.2.2-1focal.20220210.221801 of webots_ros2_importer

**Expected behavior**
Visual and collision models should be aligned in proto .

**Affected Packages**
List of affected packages:
  - webots_ros2_importer

**System**
 - Webots Version: R2022a
 - ROS Version: Galactic
 - Operating System: Ubuntu 20.04

BenjaminHug8, 2022-03-04:
Hi thank you for raising this issue, I will take a look!

BenjaminHug8, 2022-03-04:
Ok it seems that this fix https://github.com/cyberbotics/urdf2webots/pull/147 on the converter tool has been merged after the last release of `webots_ros2`, so we just need to update the converter tool for the next release.

BenjaminHug8, 2022-03-07:
A temporary solution would be to go on this [repository](https://github.com/cyberbotics/urdf2webots) and download the tool used in `webots_ros2_importer` to convert your URDF in a PROTO file.

closed 2023-04-17 by ygoumaz