cyberbotics/webots_ros2 #413: Tesla example: robot window error on `master`

BenjaminHug8, 2022-03-08:
**Describe the Bug**
I recently tested the `Tesla` example and saw that there is an error for the robot window of Webots (it seems to not work for vehicle).

**Steps to Reproduce**
1. Launch Tesla example.
2. See error:
```
[driver-3] Error: libgeneric_robot_window.so: cannot open shared object file: No such file or directory (dynamic library)
[driver-3] Error: failed to load /home/benjamin/webots_master/projects/vehicles/plugins/robot_windows/automobile/libautomobile.so library
```

**Expected behavior**
No errors.

ahartlba, 2022-08-04:
Had a similar problem when controlling vehicles with ros.

Maybe try adding the following to your bashrc, it worked for me. (I run on python 3.8)
```bash
export WEBOTS_HOME=/usr/local/webots
export LD_LIBRARY_PATH=$WEBOTS_HOME/lib/controller:$LD_LIBRARY_PATH
export PYTHONPATH=$PYTHONPATH:${WEBOTS_HOME}/lib/controller/python38
```

ygoumaz, 2023-02-22:
To fix this issue, `libgeneric_robot_window.so` should be compiled with the `libController`, `libCar` and `libDriver` in `webots_ros2_driver`.

closed 2023-05-22 by ygoumaz