cyberbotics/webots_ros2 #418: Questions about tf2_ros, it can't work

MistyMoonR, 2022-03-15:
I have created a robot inside webots, a small car with velodyne lidar. Follow the simple tutorial to finish creating the ros2 package. Some of the code is derived from `webots_ros2_turtlebot`. Finally IMU and lidar and cart cmd_vel are working fine. Thanks to webots_ros2! 

**BUG**
When I run rviz2, It will be found that the machine is moving inside the webots, but the TF is fixed on Rviz. (Frames can shows all TF)

**ERROR**
RVIZ:
> Global Status: Error             
>Frame[map] does not exist in RVIZ2

Terminal:
>Message Filter dropping message: frame 'Velodyne_VLP_16' at time 1 for reason 'Unknown'

I tried to replace `odom` but it didn't work. After careful comparison with the `ros2 topic list` of the tutorial, I found that `/odom` and `/map` are indeed missing. After my exploration, I couldn't get successfully, so please forgive my problem.

I personally feel that the key position is on this [footprint_publisher](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_turtlebot/launch/robot_launch.py#L86)      

but I don't know how to fix it. Can you tell me how to go about it?

**System**
 - Webots Version: R2022a
 - ROS Version: Foxy
 - Operating System: Linux Ubuntu 20.04 LTS
 - VMware Workstation

lukicdarkoo, 2022-03-15:
If you followed this tutorial:
https://docs.ros.org/en/foxy/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html

then `odom` and `map` frames will not be published. You have to implement at least the odometry.

MistyMoonR, 2022-03-17:
> If you followed this tutorial: https://docs.ros.org/en/foxy/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html
> 
> then `odom` and `map` frames will not be published. You have to implement at least the odometry.

YES, How to implement the odometer is exactly what I was asking.

I watched this tutorial and there is no talk about `odom`. After I tested this [webots_ros2_turtlebot](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_turtlebot) separately, using `view_frames` the output is as follows As you can see from the image, there is `/odom` above the `base_link`. I looked closely at the code and found that this seems to be automatically published.

[image]
------

However, I ran the bot I created and found that the node was not published.

[image]

For this reason, I've checked a lot of information online, but I can't find a solution to the problem. Can you tell me how to solve this problem? I would appreciate it.

lukicdarkoo, 2022-03-17:
You have to calculate and publish odometry by yourself (e.g. from wheels or IMU or camera or any other source that can produce a high-frequency pose estimation without discrete jumps). In the TurtleBot example, the [diff_drive_controller](http://control.ros.org/ros2_controllers/diff_drive_controller/doc/userdoc.html) is used to receive velocity commands and publish odometry. I suggest you take a look at the TurtleBot example and also utilize the `diff_drive_controller`.

MistyMoonR, 2022-03-18:
> You have to calculate and publish odometry by yourself (e.g. from wheels or IMU or camera or any other source that can produce a high-frequency pose estimation without discrete jumps). In the [...]

Thanks for the reply! I am trying to subscribe to `/imu` and `/gps` and publish `/odom`. These nodes are available directly inside `webots_ros2_driver`. I'm sorry I'm spoiled by commodities, because I found out that the ros package I have for the 9-axis gyro is available directly use `/odom`, so I haven't paid much attention to this aspect. It would help me a lot if you could update and provide the `/odom` node in [webots_ros2_mavic](https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_mavic). And this method will be more suitable for most of the bots. The `diff_drive_controller` only works for ground-based normal two-wheeled robots. 

Thanks again to Cyberbotics for their contribution to the open source community!

ygoumaz, 2023-02-21:
While adding odometry to the mavic example isn't currently in our plans, you are welcome to propose your own contribution! We would be more than happy to review and offer assistance with any PR you might open.

closed 2023-02-21 by ygoumaz