cyberbotics/webots_ros2 #419: webots_ros2 for universal robot with robotiq 85 gripper

svenkatachalam, 2022-03-16:
Hello all,

I am trying to simulate the object grasping with robotiq 85 gripper on the universal robot with ros2 control. I am using ROS Foxy with Ubuntu 20.04.
I have done the following:
1. Updated the hardware description urdf file
2. Included the joints in controller manager file

With the updated files in the launch file, when I tried with URDF (robot description for robot_state_publisher) generated outside of webots (including the gripper), the error is  "[driver-2]   what():  Invalid URDF, it doesn't contain a <robot> tag". I checked the URDF for any error using urdf_check, but it returned that URDF is fine.

After including the updated files in launch file, I am getting error as. I tried using generated URDF from webots, and the error remained same. When I remove the description corresponding to gripper from "controller manager yaml" and "hardware description urdf", the error is gone.

I would like to know, if it would be possible to use external grippers other than already existing (like Robotiq-3f-gripper). 
For using our own URDF, I saw from https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Ros2Supervisor-Import-your-Urdf-Robot, that we need to use Ros2supervisor to import so. But this feature is not available with foxy branch of 
[image]
[image]
[image]
[image]
webots_ros2.

ad-daniel, 2022-03-22:
From the error it appears that it cannot find the robot tag, but from the screenshot it seems to be there. Can you provide the entire URDF ?

svenkatachalam, 2022-03-23:
Hello Mr.Daniel, Thank you. I have attached the URDF files (Robot URDF and Hardware URDF). When I export URDF from Webots, the ordering of links and joints are different (in gripper), so I reorder them in the Robot URDF. I have attached the three URDF files here.
[URDF.tar.gz](https://github.com/cyberbotics/webots_ros2/files/8330643/URDF.tar.gz)

svenkatachalam, 2022-03-24:
@ad-daniel : I found the hard-way  that the robot should have the supervisor "true", to have efficient control. Now controlling the robot and the gripper with ros2 is working fine. I will close the issue. Thank you for your help.