cyberbotics/webots_ros2 #420: Loading a dynamic Pedestrian node to ROS2 webots world

stef264, 2022-03-21:
**Describe the Bug**
I have a bug in my personal world adding a dynamic Pedestrian node and letting it move.
It is possible to reproduce the error with the *webots_ros2_mavic* package.

**Steps to Reproduce**
1. Add dynamic Pedestrian node to *mavic_world.wbt*
```
Pedestrian {
  translation -5.22 0 1.27
    controllerArgs [
    "--trajectory= -3 0, 3 0"
    "--speed=2.5"
  ]
}

```

After running `ros2 launch webots_ros2_mavic robot_launch.py` i get following console output:
```
INFO: Starting extern controller for robot "Mavic 2 PRO".
INFO: pedestrian: Starting controller: python3 -u pedestrian.py "--trajectory= -3 0, 3 0" --speed=2.5
Traceback (most recent call last):
  File "pedestrian.py", line 16, in <module>
    from controller import Supervisor
  File "/usr/local/webots/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: /usr/local/webots/lib/controller/python38/_controller.so: undefined symbol: _ZN6webots5Robot19internalGetInstanceEv
WARNING: 'pedestrian' controller exited with status: 1.
```

There is no error if i remove the \<extern> Controller and run Webots without ROS2.

**System**
 - Webots Version: R2022a
 - ROS Version: Galactic
 - Operating System: Ubuntu 20.04
 
**Additional context**
Is this some ROS2 <-> Webots specific error or is something wrong with Webots?
Maybe this issue needs to be moved to the Webots issue page.

lukicdarkoo, 2022-03-22:
Internally, libController libraries deployed with webots_ros2 and Webots are slightly different and they are both present in `LD_LIBRARY_PATH`. Therefore, Python bindings get confused. To solve it you can:
```py
if os.getenv('ROS_DISTRO') is not None:
    from webots_ros2_driver_webots.controller import Supervisor
else:
    from controller import Supervisor
```

stef264, 2022-03-23:
Works great! Thanks.

Is this classified as a bug? Where do need to look into, to fix it?

closed 2023-07-26 by ygoumaz