cyberbotics/webots_ros2 #422: Ros2 LIDAR data structure definition does not match data

sebastianohl, 2022-03-26:
**Describe the Bug**
The lidar plugin in the webots ros2 driver outputs data as sensor_msg.pointcloud2 data structure. This data structure have an array how the binary data is structured. the plugin describes the data as 3 float values (x,y,z). Unfortunately this is not according to the webots lidar point data structure WbLidarPoint[].

**Steps to Reproduce**
1. mount a Sick LMS291 lidar sensor to a vehicle
2. enable ros output
3. receive pointcloud2 data in a ros2 node and try to decode it

**Expected behavior**
the expected behavior is that the data in described in message.fields should match the data stored in message.data. The data described by the datastructure WbLidarPoint (https://cyberbotics.com/doc/reference/lidar?tab-language=c). It does not only contain x,y,z as floats but also the layer (int32) and the time(float32). This need to be added to the fields array.

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**System**
 - Webots Version: R2022a
 - ROS Version: galactic
 - Operating System: Ubuntu 20.04
 - Graphics Card: vmware

omichel, 2022-05-25:
@ad-daniel: was this issue fully addressed in #4545? If so, can we close it?

ad-daniel, 2022-05-25:
Not yet, only on the ROS1 side at the moment