cyberbotics/webots_ros2 #423: Build on Mac studio M1 Max

zegangYang, 2022-03-27:
**Describe the Bug**
build got link error 
**Steps to Reproduce**
1. `source ~/ros2_galatic/install/local_setup.zsh`
2. follow build on macOS steps but use these settings
```
colcon build \
  --symlink-install \
  --merge-install \
  --event-handlers console_cohesion+ console_package_list+ \
  --packages-skip-build-finished \
  --cmake-args \
    --no-warn-unused-cli \
    -DBUILD_TESTING=OFF \
    -DINSTALL_EXAMPLES=ON \
    -DCMAKE_OSX_SYSROOT=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk \
    -DCMAKE_OSX_ARCHITECTURES="arm64"
```
4. 
5. See error

**Expected behavior**
Build and link properly

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**Screenshots**
xx try to link x86_64 (no symbols of arm64)
**System**
 - Webots Version: R2022a
 - ROS Version: galatic
 - Operating System: macOS Monterey
 - Graphics Card: Apple Internal GPU

**Additional context**
Add any other context about the problem here.

zegangYang, 2022-03-28:
## build webots-libcontroller
### clone webots repo
```
git clone https://github.com/cyberbotics/webots.git
cd webots.git
git submodule init
git submodule update --recursive
```
### make some changes of Makefile
*since QtWebEngine disabled when install qt@5 from brew , we actually only need arm64  webots-libcontroller dylib*
*so skip webots qt project*
```
diff --git a/Makefile b/Makefile
index d0bf78133..8f8d6b29b 100644
--- a/Makefile
+++ b/Makefile
@@ -116,11 +116,11 @@ endif
        @+echo "#"; echo "# * wren *"; echo "#"
        @+make --silent -C src/wren $(TARGET)
        @+echo "#"; echo "# * webots (core) *"; echo "#"
-       @+make --silent -C src/webots $(TARGET)
+#      @+make --silent -C src/webots $(TARGET)
        @+echo "#"; echo "# * controller library *"; echo "#"
        @+make --silent -C src/controller $(TARGET) WEBOTS_HOME="$(WEBOTS_HOME)"
        @+echo "#"; echo "# * resources *";
-       @+make --silent -C resources $(MAKECMDGOALS) WEBOTS_HOME="$(WEBOTS_HOME)"
+#      @+make --silent -C resources $(MAKECMDGOALS) WEBOTS_HOME="$(WEBOTS_HOME)"
        @+echo "#"; echo "# * projects *";
        @+make --silent -C projects $(TARGET) WEBOTS_HOME="$(WEBOTS_HOME)"
```
### build
`make -j2`

## copy libraries 
`cp webots/lib/controller/*.dylib  webots_ros2_driver/webots/lib/darwin19/`

## build webots_ros2
### quick fix PythonLibs error(python3.9)
```
diff --git a/webots_ros2_driver/CMakeLists.txt b/webots_ros2_driver/CMakeLists.txt
index a803306..a2f7e87 100644
--- a/webots_ros2_driver/CMakeLists.txt
+++ b/webots_ros2_driver/CMakeLists.txt
@@ -17,7 +17,10 @@ find_package(vision_msgs REQUIRED)
 find_package(webots_ros2_msgs REQUIRED)
 find_package(tinyxml2_vendor REQUIRED)
 find_package(TinyXML2 REQUIRED)
-find_package(PythonLibs 3.8 EXACT REQUIRED)
+find_package(PythonInterp REQUIRED)
+find_package(PythonLibs ${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR} EXACT REQUIRED)
+find_package(Python3 REQUIRED COMPONENTS Development NumPy)
+

 if (UNIX AND NOT APPLE)
   set(WEBOTS_LIB_BASE webots/lib/linux-gnu)
@@ -112,9 +115,10 @@ ament_target_dependencies(driver
   tinyxml2_vendor
   TinyXML2
 )
+
 target_link_libraries(driver
   ${WEBOTS_LIB}
-  ${PYTHON_LIBRARIES}
+  ${Python3_LIBRARIES}
 )
 install(
   DIRECTORY include/
```

### quick fix some lost python packages
```
python3 -m pip install pyclibrary
python3 -m pip install netifaces
```

ygoumaz, 2022-12-29:
I am closing this issue, as the ROS distribution is outdated, the libcontroller is now fully compiled during the build process and the macOS workflow now depends on a UTM Virtual machine.

closed 2022-12-29 by ygoumaz