cyberbotics/webots_ros2 #424: Tiago example:Slam can't work

discleave, 2022-03-29:
**Describe the Bug**
'TIAGo SLAM' in 'TIAGo Example' can't build the map

**Steps to Reproduce**
1. execute 'ros2 launch webots_ros2_tiago robot_launch.py rviz:=true slam:=true'
2. See error 'No map recieived' in rviz
3. execute 'ros2 run teleop_twist_keyboard teleop_twist_keyboard'
4. See error. I can control TIAGo by keyboard, but the map can't be built

**Expected behavior**
I could control TIAGo to build a map.

**Screens
[image]
[image]

**System**
 - Webots Version: 2022a
 - ROS Version:foxy
 - Operating System: Ubuntu 20.04.3

ad-daniel, 2022-03-29:
Personally it appears to be working for me, the map topic is published for me, but I do have the LaserRangeScan warning you mention. Did you follow the install procedure described here? https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install

ygoumaz, 2023-02-17:
I am closing this issue. Feel free to reopen if needed.

closed 2023-02-17 by ygoumaz