cyberbotics/webots_ros2 #427: Different drivers are not correctly binded to the corresponding robots

devis12, 2022-04-03:
**Describe the Bug**
I have a world with two different robots: one called '**gantry**' and the other '**my_simple_robot2**'. In their init method I retrieved with `getDevice()` the reference of some of their motors, so that I can use them in the periodic step call. I've specified in the urdf files and in the node parameters the name of the robot in the .wbt definition with '`webots_robot_name`', but not all the runs perform the correct binding at run time. In the launch params within /tmp I can see that the correct urdf are attached to the corresponding driver which should manage the robot, but in the log you can easily see the difference: e.g. in some runs there is a log I've put at the end of the init method which confirms the init has performed correctly, in others a lot of errors are displayed within the step function, because I'm using the references to the motors which were not found (you can comment them out) and they were not found because an incorrect binding was made. For example, I use the '`2`' at the end of 'my_simple_robot2' (the robot name), to retrieve the reference to its left wheel motor which is called '`left_wheel_motor_`' + '`2`' (with the two retrieved from above), but in multiple runs I got `[driver-3] Warning: "left_wheel_motor_y" device not found`, suggesting me that instead of the 2 at the end of name I have the 'y' of 'gantry' (which is the name of the other robot!). If references are messed up, it means that the binding has been performed uncorrectly, hence I'm not working with the motors I'm supposed to and all their references are basically null. I know that the issue can be solved with a single driver class and if branch depending on the type of robot I'm working with, but it seems way less clean as a solution.
I hope I'm not doing anything wrong, I'll link the package to reproduce the errors, cleaned up from things that does not concern with the error as much as possible. Hope to receive a suggestion or fix from you. Thank you in advance.

**Steps to Reproduce**
1. Go to https://github.com/devis12/webots_ros2_show_issue: it's a ros2 package that you can easily compile in your ws with colcon build
2. source the `install/local_setup.bash` of the workspace
3. Give the command `ros2 launch webots_ros2_show_issue blocks_world.launch.py`
4. Rerun it a couple of times: you'll see sometimes errors because references are messed up (e.g. you get something like `[driver-3] Warning: "left_wheel_motor_y" device not found.`), while in other cases the binding will be performed correctly (e.g. in the logs you see `[driver-3] [INFO] [1648990034.986132908] [my_simple_robot2_driver]: my_simple_robot2 driver support node set up`)

**Expected behavior**
Since what I've specified in the urdf and launch files, I'd expect no race condition of any kind and the binding to the correct controller should be always correct!

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

I guess only that especially the PythonPlugin class where you retrieve the python driver instance.

**Screenshots**
[image]
Same launch file run: first time motors are not found, second time they are!

**System**
 - Webots Version: R2022a 17/12/2021
 - ROS Version: Foxy
 - Operating System: Linux Mint 20.3 Cinnamon (Ubuntu Focal)
 - Graphics Card: NVIDIA geforce gtx 850m

**Additional context**
Give a look at the urdf and world files, if I missed something, but in case I did, it means that it's not specified in your wiki or examples either. In any case, I feel like your demos are missing a case like this: multiple robots (better if connected as here to different controllers).

lukicdarkoo, 2022-04-03:
When having multiple robots you need to define the `WEBOTS_ROBOT_NAME` environment variable to correctly assign controllers, see:
https://github.com/cyberbotics/webots_ros2/blob/614b533b987e8838953ba85da50c0e690e93eff5/webots_ros2_universal_robot/launch/multirobot_launch.py#L49

devis12, 2022-04-03:
Thank you a lot for the quick answer.

closed 2022-04-03 by devis12