cyberbotics/webots_ros2 #439: ros2 cpp package not working

CharlesXu1124, 2022-04-30:
I was following the tutorial on https://docs.ros.org/en/foxy/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html and try to create a c++ package in order to load my custom robot. Everything works fine but I am stuck at step 4, in which I was supposed to create a .urdf file that points to my compiled webots driver, but then when I try to run the ros2 package after building, it was not able to find that driver file. On the other hand, I was able to manually run the driver file and it starts to subscribe to my ros2 messages, but not when I launch the ros2 package. Does anyone know what is happening?

MistyMoonR, 2022-05-04:
Perhaps you should check the `CMakeLists.txt` or `setup.py` file to see if the URDF has been imported correctly. 

`CMakeLists.txt`
``` c++
install(
  DIRECTORY resource
  DESTINATION share/${PROJECT_NAME}
)
```
`setup.py`
```python
data_files.append(('share/' + package_name + '/resource', ['resource/*.urdf']))
```

ygoumaz, 2023-03-02:
In the tutorial, it is mentioned at [step 6](https://docs.ros.org/en/foxy/Tutorials/Advanced/Simulators/Webots.html#modify-the-setup-py-file) that you have to add the urdf file in the `setup.py` script so that it get importe to your package after build and installation.

In the case of C++, this should be done inside the `CMakeLists.txt` file, as @MistyMoonR said. You can find more informations on how to use this file in the tutorial about C++ plugins [here](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Creating-a-Custom-Cpp-Plugin).

SenHe524, 2023-03-03:
> 

I have successfully run the function pack of C++ to co-simulate with webots. Thank you very much!

closed 2023-03-03 by ygoumaz