cyberbotics/webots_ros2 #441: dll library cannot be loaded

wangruoyao, 2022-05-08:
Tesla and other demo cannot work properly.

For example, Tesla can follow the lane at first but then bump into the wall. The Mavic cannot  be controlled.

 - Webots Version: R2022a
 - ROS Version: foxy
 - Operating System: Windows 10
 - Graphics Card: GTX1070Ti

Command's feedback is here:

[driver.EXE-3] Error: ????????????顣
[driver.EXE-3]  (dynamic library)
[driver.EXE-3] Error: failed to load C:/Program Files/Webots/projects/vehicles/plugins/robot_windows/automobile/automobile.dll library
[webots-1]
[webots-1] ODE Message 3: LCP internal error, s <= 0 (s=0.0000e+00)
[webots-1]

[driver.EXE-2] Error: ????????????顣
[driver.EXE-2]  (dynamic library)
[driver.EXE-2] Error: failed to load C:/Program Files/Webots/resources/projects/plugins/robot_windows/generic/generic.dll library

I thought it lack system environment variable and checked it, but found it already existed.
So, any useful information?

omichel, 2022-05-09:
Does it help if you add `C:/Program Files/Webots/projects/vehicles/plugins/robot_windows/automobile;C:/Program Files/Webots/resources/projects/plugins/robot_windows/generic` in your PATH environment variable?

wangruoyao, 2022-05-10:
> This code with the Tesla vehicle is not meant to be robust. It was provided as a simple demo to get people started with the ROS 2 interface. It is indeed normal that it doesn't behave very well.

Got it. Although it shows that it lacks dll，but looks like this error doesn't affect the behave. And the Mavic cannot be controlled is probably that I didn't install the control relative package. I'll try again once installed the package.

closed 2022-07-08 by wangruoyao