cyberbotics/webots_ros2 #442: how to build webots_ros2_driver with brew version python?

xlla, 2022-05-14:
**Describe the Bug**
WebotsNode failure to launch.

First , I have download webots R2022a,  but encounter webots issue [#4528](https://github.com/cyberbotics/webots/issues/4528)

Then I have installed webots night-build, after build webots_ros2_driver and try the [guide](https://docs.ros.org/en/humble/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html) , 

It failure with exit code -11
<pre>
[ERROR] [driver-2]: process has died [pid 63124, exit code -11, cmd '/Users/xlla/ros2_dev_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /var/folders/6j/gqhq_r1n0wg5qjypvtkccnqr0000gn/T/launch_params_6lf31amf'].
</pre>

Then I checkout webots source a90eb4e59654bcd8d8ebe9de0c238426ec489576, 
I am not understand exactly the python dependencies, so I followed [this dependencies guide](https://github.com/cyberbotics/webots/wiki/macOS-Optional-Dependencies) installed python 3.7, 3.8, 3.9, 3.10 from python.org and from brew ,  and built it, 
then link the WEBOTS_HOME/lib/controller to webots_ros2/webots_ros2_driver/webots/lib/darwin19
 and manual create __init__.py inside python39

after build webots_ros2_driver and try the [guide](https://docs.ros.org/en/humble/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html) , 

It failure with exit code -6
<pre>
slim:ros2_dev_ws xlla$ ros2 launch my_package robot_launch.py
[INFO] [launch]: All log files can be found below /Users/xlla/.ros/log/2022-05-14-18-33-07-347218-slim.local-80582
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots-1]: process started with pid [80583]
[INFO] [driver-2]: process started with pid [80584]
[driver-2] Webots doesn't seems to be ready yet: (retry count 1)
[driver-2] Cannot open file: /var/tmp/webots-80586/WEBOTS_SERVER (retry count 2)
[driver-2] Cannot open file: /var/tmp/webots-80586/WEBOTS_SERVER (retry count 3)
[driver-2] Cannot open file: /var/tmp/webots-80586/WEBOTS_SERVER (retry count 4)
[driver-2] libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Error: The Python module with the WebotsNode class cannot be executed.
[ERROR] [driver-2]: process has died [pid 80584, exit code -6, cmd '/Users/xlla/ros2_dev_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /var/folders/6j/gqhq_r1n0wg5qjypvtkccnqr0000gn/T/launch_params_cgzp0880'].
</pre>

1. Since I have mainly work on brew python3.9, I have changed CMakeLists.txt to use 3.9 instead of 3.10, does it must be 3.10?
2. I noticed each controller python lib ( _controller.so ) was linked with  different version python lib,  to match with my work python version, I have change CMakeLists.txt as this
<pre>
  ament_python_install_package(${PROJECT_NAME}_webots
    PACKAGE_DIR ${WEBOTS_LIB_BASE}/python39_brew)
</pre>
it still error with exit code -6

3. I have noticed that all  python*_brew folder missing _vehicle.so and vehicle.py , is this the reason?
 

**Steps to Reproduce**
1. follow [guide](https://docs.ros.org/en/humble/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html)
2.  `ros2 launch my_package robot_launch.py`

**Expected behavior**
WebotsNode successful launch.

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**Screenshots**
<pre>
slim:ros2_dev_ws xlla$ ros2 launch my_package robot_launch.py
[INFO] [launch]: All log files can be found below /Users/xlla/.ros/log/2022-05-14-18-33-07-347218-slim.local-80582
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots-1]: process started with pid [80583]
[INFO] [driver-2]: process started with pid [80584]
[driver-2] Webots doesn't seems to be ready yet: (retry count 1)
[driver-2] Cannot open file: /var/tmp/webots-80586/WEBOTS_SERVER (retry count 2)
[driver-2] Cannot open file: /var/tmp/webots-80586/WEBOTS_SERVER (retry count 3)
[driver-2] Cannot open file: /var/tmp/webots-80586/WEBOTS_SERVER (retry count 4)
[driver-2] libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Error: The Python module with the WebotsNode class cannot be executed.
[ERROR] [driver-2]: process has died [pid 80584, exit code -6, cmd '/Users/xlla/ros2_dev_ws/install/webots_ros2_driver/lib/webots_ros2_driver/driver --ros-args --params-file /var/folders/6j/gqhq_r1n0wg5qjypvtkccnqr0000gn/T/launch_params_cgzp0880'].
</pre>

**System**
 - Webots Version: [ R2020a a90eb4e59654bcd8d8ebe9de0c238426ec489576]
 - ROS Version: [humble] 
 - Operating System: [macOS High sierra]
 - Graphics Card: [NVIDIA GeForce GTX 1060 6 GB]

**Additional context**

I have installed many pip packages include ros2 under python 3.9, so I rather stick to this version.
My ros2 is built from source, it works for all tutorials of ros2 up to [this](https://docs.ros.org/en/humble/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html) .
My webots is built from source, it works for [this](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-Create-Webots-Robot)
I can run python version controller examples, so I think the controller lib is fine.
I can create a simple ros2 subscriber node to manual initial a Robot from controller and driver a bot insider webots world.

but failure to launch the webotsNode as this [guide](https://docs.ros.org/en/humble/Tutorials/Simulators/Webots/Setting-up-a-Robot-Simulation-Webots.html)

omichel, 2022-05-16:
#4528 should be resolved by now. Please note that [Humble only has tier 3 support for macOS](https://docs.ros.org/en/humble/Releases/Release-Humble-Hawksbill.html).

wuyue92tree, 2022-07-07:
I also got this issue, want help~

System:
- Webots Version: [ R2020a]
- Python Version: [3.8.13]
- ROS Version: [foxy]
- Operating System: [macOS High sierra]
- Graphics Card: [Radeon Pro 560 4 GB]

alexandrosnic, 2022-09-22:
Hey @xlla , I encountered this error as well in Linux. Did you solve it?

nageng2, 2023-01-19:
in my case...
remove every environment variables for webots in .bashrc (WEBOTS_HOME, ${WEBOTS_HOME}/lib/controller)
and resolve it...

ygoumaz, 2023-03-02:
I am closing this issue now, as we now only support the package in a Linux environment (i.e. in a UTM Virtual Machine on macOS devices).

Feel free to reopen if the problem occurs in Linux.

closed 2023-03-02 by ygoumaz