cyberbotics/webots_ros2 #443: webots_ros2 install problem

adminzone24, 2022-05-23:
Describe the Bug
Failed   <<< webots_ros2_driver [0.40s, exited with code 2]

Starting >>> webots_ros2_msgs
Starting >>> webots_ros2_importer
Finished <<< webots_ros2_msgs [0.41s]                                     
Starting >>> webots_ros2_driver
Starting >>> webots_ros2_core
--- stderr: webots_ros2_importer                                    
package init file 'webots_ros2_importer/urdf2webots/urdf2webots/__init__.py' not found (or not a regular file)
---
Finished <<< webots_ros2_importer [0.56s]
--- stderr: webots_ros2_driver                                              
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libController.so when searching for -lController
/usr/bin/ld: cannot find -lController
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppController.so when searching for -lCppController
/usr/bin/ld: cannot find -lCppController
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libdriver.so when searching for -ldriver
/usr/bin/ld: cannot find -ldriver
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppDriver.so when searching for -lCppDriver
/usr/bin/ld: cannot find -lCppDriver
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libcar.so when searching for -lcar
/usr/bin/ld: cannot find -lcar
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppCar.so when searching for -lCppCar
/usr/bin/ld: cannot find -lCppCar
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/webots_ros2_driver_imu.dir/build.make:202: libwebots_ros2_driver_imu.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/webots_ros2_driver_imu.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libController.so when searching for -lController
/usr/bin/ld: cannot find -lController
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppController.so when searching for -lCppController
/usr/bin/ld: cannot find -lCppController
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libdriver.so when searching for -ldriver
/usr/bin/ld: cannot find -ldriver
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppDriver.so when searching for -lCppDriver
/usr/bin/ld: cannot find -lCppDriver
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libcar.so when searching for -lcar
/usr/bin/ld: cannot find -lcar
/usr/bin/ld: skipping incompatible /home/su/webots_ros2/src/webots_ros2/webots_ros2_driver/webots/lib/linux-gnu/libCppCar.so when searching for -lCppCar
/usr/bin/ld: cannot find -lCppCar
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/driver.dir/build.make:370: driver] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/driver.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< webots_ros2_driver [0.40s, exited with code 2]
Aborted  <<< webots_ros2_core [0.50s]                      

Summary: 2 packages finished [1.03s]
  1 package failed: webots_ros2_driver
  1 package aborted: webots_ros2_core
  2 packages had stderr output: webots_ros2_driver webots_ros2_importer

List of affected packages:
webots_ros2_driver

System
 - Webots Version: R2020a
 - ROS Version: foxy
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: Macbook M1

ad-daniel, 2022-05-25:
Hi, which branch are you trying to build? master or develop? Also do you have urdf2webots installed separately already? (for instance from pip)

ad-daniel, 2022-05-25:
It might have been due to a synchronization issue on our end which has been fixed, could you try again and report back if it solves your issue?

ad-daniel, 2022-05-25:
> Does this problem have anything to do with the computer system? Because I use mackbook M1.

Likely yes since webots_ros2_driver relies on libcontroller provided by webots which currently only generates x86_64 libraries, whereas the M1 needs arm64. We are in the process of adding native support of M1 here https://github.com/cyberbotics/webots/pull/4543, which I believe will be supported soon (next week, more or less)

omichel, 2022-05-25:
Note: native M1 support is only for macOS, not Linux.

ygoumaz, 2022-12-29:
Since #508, the libController is compiled directly when building the package, allowing ARM compatibility. This issue is now fixed.

closed 2022-12-29 by ygoumaz