cyberbotics/webots_ros2 #451: Colcon build: Failed   <<< webots_ros2_control

Fran-FC, 2022-06-08:
**Describe the Bug**
The build fails despite following all the steps

**Steps to Reproduce**
1. Followed the guide: https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install 

**Expected behavior**
It should compile

**Affected Packages**
List of affected packages:
  - webots_ros2_control

**Screenshots**
Starting >>> webots_ros2_control
--- stderr: webots_ros2_control                               
In file included from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2ControlSystem.cpp:15:
/home/fran/ros_ws/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:77:37: error: ‘hardware_interface::return_type webots_ros2_control::Ros2ControlSystem::read(const rclcpp::Time&, const rclcpp::Duration&)’ marked ‘override’, but does not override
   77 |     hardware_interface::return_type read(const rclcpp::Time &/*time*/, const rclcpp::Duration &/*period*/) override;
      |                                     ^~~~
/home/fran/ros_ws/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:78:37: error: ‘hardware_interface::return_type webots_ros2_control::Ros2ControlSystem::write(const rclcpp::Time&, const rclcpp::Duration&)’ marked ‘override’, but does not override
   78 |     hardware_interface::return_type write(const rclcpp::Time &/*time*/, const rclcpp::Duration &/*period*/) override;
      |                                     ^~~~~
/home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp: In member function ‘virtual void webots_ros2_control::Ros2Control::step()’:
/home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:46:31: error: no matching function for call to ‘controller_manager::ControllerManager::read(rclcpp::Time, const rclcpp::Duration&)’
   46 |       mControllerManager->read(mNode->get_clock()->now(), dt);
      |       ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2Control.hpp:32,
                 from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:15:
/opt/ros/humble/include/controller_manager/controller_manager.hpp:134:8: note: candidate: ‘void controller_manager::ControllerManager::read()’
  134 |   void read();
      |        ^~~~
/opt/ros/humble/include/controller_manager/controller_manager.hpp:134:8: note:   candidate expects 0 arguments, 2 provided
/home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:59:30: error: no matching function for call to ‘controller_manager::ControllerManager::write(rclcpp::Time, const rclcpp::Duration&)’
   59 |     mControllerManager->write(mNode->get_clock()->now(), dt);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2Control.hpp:32,
                 from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:15:
/opt/ros/humble/include/controller_manager/controller_manager.hpp:141:8: note: candidate: ‘void controller_manager::ControllerManager::write()’
  141 |   void write();
      |        ^~~~~
/opt/ros/humble/include/controller_manager/controller_manager.hpp:141:8: note:   candidate expects 0 arguments, 2 provided
In file included from /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp:57,
                 from /opt/ros/humble/include/class_loader/class_loader/class_loader.hpp:55,
                 from /opt/ros/humble/include/class_loader/class_loader/multi_library_class_loader.hpp:52,
                 from /opt/ros/humble/include/pluginlib/pluginlib/class_loader.hpp:38,
                 from /home/fran/ros_ws/install/webots_ros2_driver/include/webots_ros2_driver/WebotsNode.hpp:30,
                 from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:33,
                 from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2ControlSystem.cpp:15:
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = webots_ros2_control::Ros2ControlSystem; B = webots_ros2_control::Ros2ControlSystemInterface]’:
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:216:7:   required from here
/opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:218:12: error: invalid new-expression of abstract class type ‘webots_ros2_control::Ros2ControlSystem’
  218 |     return new C;
      |            ^~~~~
In file included from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2ControlSystem.cpp:15:
/home/fran/ros_ws/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:56:9: note:   because the following virtual functions are pure within ‘webots_ros2_control::Ros2ControlSystem’:
   56 |   class Ros2ControlSystem : public Ros2ControlSystemInterface
      |         ^~~~~~~~~~~~~~~~~
In file included from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2ControlSystem.hpp:27,
                 from /home/fran/ros_ws/src/webots_ros2/webots_ros2_control/src/Ros2ControlSystem.cpp:15:
/opt/ros/humble/include/hardware_interface/system_interface.hpp:169:23: note:     ‘virtual hardware_interface::return_type hardware_interface::SystemInterface::read()’
  169 |   virtual return_type read() = 0;
      |                       ^~~~
/opt/ros/humble/include/hardware_interface/system_interface.hpp:178:23: note:     ‘virtual hardware_interface::return_type hardware_interface::SystemInterface::write()’
  178 |   virtual return_type write() = 0;
      |                       ^~~~~
gmake[2]: *** [CMakeFiles/webots_ros2_control_system.dir/build.make:76: CMakeFiles/webots_ros2_control_system.dir/src/Ros2ControlSystem.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/webots_ros2_control_system.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/webots_ros2_control.dir/build.make:76: CMakeFiles/webots_ros2_control.dir/src/Ros2Control.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/webots_ros2_control.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< webots_ros2_control [6.92s, exited with code 2]

Summary: 6 packages finished [32.6s]
  1 package failed: webots_ros2_control
  5 packages had stderr output: webots_ros2_control webots_ros2_core webots_ros2_importer webots_ros2_mavic webots_ros2_tesla
  6 packages not processed

**System**
 - Webots Version: R2020a snap version
 - ROS Version: ROS2 humble 
 - Operating System: Ubuntu 22.04 LTS

ad-daniel, 2022-06-08:
Hi, are you building from the master branch? Did you install humble from the main repository or testing?

Fran-FC, 2022-06-08:
Hi! I installed humble from the debs, using the official guide https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html (the main repo I guess) 

I am building from the master branch.

Thanks for the reply :)

Fran-FC, 2022-06-08:
I changed the branch and it compiles now. Everything seems to work fine. Thank you for the support!

closed 2022-06-08 by ad-daniel