cyberbotics/webots_ros2 #455: Textures don't show up in example ros2 launch files

royito55, 2022-06-24:
**Describe the Bug**
I installed webots R2022a through snap and then webots_ros2 through apt. When I launch the simulations, it seems like webots can't download some resources it needs:

`WARNING: DEF POLE Solid > DEF POLE_SHAPE Shape > CorrodedMetal (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/corroded_metal/corroded_metal_metalness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/corroded_metal/corroded_metal_metalness.jpg
WARNING: DEF POLE Solid > DEF POLE_SHAPE Shape > CorrodedMetal (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/corroded_metal/corroded_metal_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/corroded_metal/corroded_metal_normal.jpg
WARNING: WoodenPalletStack (PROTO) > WoodenPallet (PROTO) > Transform > Shape > RoughPine (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/rough_pine/rough_pine_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/rough_pine/rough_pine_base_color.jpg
WARNING: WoodenPalletStack (PROTO) > WoodenPallet (PROTO) > Transform > Shape > RoughPine (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/rough_pine/rough_pine_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/rough_pine/rough_pine_roughness.jpg
WARNING: WoodenPalletStack (PROTO) > WoodenPallet (PROTO) > Transform > Shape > RoughPine (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/rough_pine/rough_pine_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/rough_pine/rough_pine_normal.jpg
WARNING: WoodenPalletStack (PROTO) > WoodenPallet (PROTO) > Transform > Shape > RoughPine (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/rough_pine/rough_pine_occlusion.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/rough_pine/rough_pine_occlusion.jpg
WARNING: ConstructionLamp (PROTO) > Shape > ReflectiveSurface (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/reflective_surface/reflective_surface_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/reflective_surface/reflective_surface_roughness.jpg
WARNING: ConstructionLamp (PROTO) > Shape > ReflectiveSurface (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/reflective_surface/reflective_surface_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/reflective_surface/reflective_surface_normal.jpg
WARNING: ConstructionLamp (PROTO) > MetalPipePaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/metal_pipe_paint/metal_pipe_paint_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/metal_pipe_paint/metal_pipe_paint_base_color.jpg
WARNING: ConstructionLamp (PROTO) > MetalPipePaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/metal_pipe_paint/metal_pipe_paint_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/metal_pipe_paint/metal_pipe_paint_roughness.jpg
WARNING: ConstructionLamp (PROTO) > MetalPipePaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/metal_pipe_paint/metal_pipe_paint_metalness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/metal_pipe_paint/metal_pipe_paint_metalness.jpg
WARNING: ConstructionLamp (PROTO) > MetalPipePaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/metal_pipe_paint/metal_pipe_paint_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/metal_pipe_paint/metal_pipe_paint_normal.jpg
WARNING: PlasticCrate (PROTO) > Transform > Shape > MattePaint (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/matte_car_paint/matte_car_paint_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/matte_car_paint/matte_car_paint_roughness.jpg
WARNING: TexturedBackground (PROTO): Cannot retrieve 'https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/factory_back.png': Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/factory_back.png: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/factory_back.png
WARNING: TexturedBackground (PROTO): Cannot download leftIrradianceUrl: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/factory_back.hdr: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/factory_back.hdr`

**Steps to Reproduce**
ros2 launch webots_ros2_universal_robot multirobot_launch.py
ros2 launch webots_ros2_tesla robot_launch.py

**Screenshots**
[image]
[image]

**System**
 - Webots Version: R2022a
 - ROS Version: Humble 
 - Operating System: Ubuntu 22.04
 - Graphics Card: Mesa Intel® Xe Graphics (TGL GT2)

royito55, 2022-06-27:
This was fixed by installing the latest nightly build deb (2022b!). However, I have a new issue with the WEBOTS_SERVER not being created in the tmp folder. It is the same as this issue:
https://github.com/cyberbotics/webots_ros2/issues/182

omichel, 2022-06-28:
In Webots R2022b, the `WEBOTS_SERVER` file is not created any more due to a change in the way the extern controllers work. The new way extern controllers work is documented [here](https://cyberbotics.com/doc/guide/running-extern-robot-controllers?version=develop). The new system is hopefully simpler and more powerful as you can launch a controller program on a different machine as the one running Webots. I am closing the issue for now as I believe it is resolved. Feel free to re-open or open a new one if needed.

closed 2022-06-28 by omichel