cyberbotics/webots_ros2 #458: Failed to build webots_ros2_driver as Required is exact version "3.10"

wangruoyao, 2022-07-08:
I got the issue when using colcon build in windows with ros2-humble
looks like the cmakelist set an exact version. How can I fix it?

Finished <<< webots_ros2_core [1.75s]
Starting >>> webots_ros2_mavic
Starting >>> webots_ros2_tesla
--- stderr: webots_ros2_driver
CMake Error at C:/Program Files/CMake/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
  (Required is exact version "3.10")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.23/Modules/FindPythonLibs.cmake:310 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:34 (find_package)

---
Failed   <<< webots_ros2_driver [2.58s, exited with code 1]
Aborted  <<< webots_ros2_tesla [0.66s]
Aborted  <<< webots_ros2_mavic [0.91s]
Aborted  <<< controller_interface [8.28s]

Summary: 13 packages finished [20.8s]
  1 package failed: webots_ros2_driver
  3 packages aborted: controller_interface webots_ros2_mavic webots_ros2_tesla
  3 packages had stderr output: webots_ros2_core webots_ros2_driver webots_ros2_importer
  11 packages not processed

wangruoyao, 2022-07-20:
I changed the CMakelist in src and compiled again
but met another issue:

Finished <<< webots_ros2_tesla [3.39s]
Finished <<< control_toolbox [31.4s]
Finished <<< webots_ros2_driver [32.1s]
Finished <<< hardware_interface [42.8s]
Starting >>> controller_interface
Finished <<< controller_interface [29.9s]
Starting >>> controller_manager
Failed   <<< controller_manager [22.0s, exited with code 1]

Summary: 17 packages finished [2min 20s]
  1 package failed: controller_manager
  4 packages had stderr output: webots_ros2_core webots_ros2_importer webots_ros2_mavic webots_ros2_tesla
  10 packages not processed

ad-daniel, 2022-07-20:
I don't have windows to test, but did you try installing python 3.10? Otherwise which version of python are you using?

wangruoyao, 2022-07-21:
> I don't have windows to test, but did you try installing python 3.10? Otherwise which version of python are you using?

I‘m using python3.83 which the humble documentation recommended.

[Install Python](https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html#id6)[](https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html#install-python)

'Open a Command Prompt and type the following to install Python via Chocolatey:

choco install -y python --version 3.8.3'

omichel, 2022-07-25:
Can you check the error message you are getting in the log file of the failing package (control_manager)? It could be a missing python module.

wangruoyao, 2022-07-26:
> Can you check the error message you are getting in the log file of the failing package (control_manager)? It could be a missing python module.

I found this in stdout.log but don't know if it is helpful.

`C:\dev_ws\src\ros2_control\controller_manager\src\realtime.cpp(17,10): fatal error C1083: Cannot open include file: 'sched.h': No such file or directory [C:\dev_ws\build\controller_manager\controller_manager.vcxproj]
  gmock.vcxproj -> C:\dev_ws\build\controller_manager\gmock\Release\gmock.lib`

omichel, 2022-07-27:
That probably means that the `ros2_control` package has problems on Windows. It doesn't seem to be related with Webots.

wangruoyao, 2022-07-28:
> That probably means that the `ros2_control` package has problems on Windows. It doesn't seem to be related with Webots.

ok, I will back to foxy.

wangruoyao, 2022-08-06:
> That probably means that the `ros2_control` package has problems on Windows. It doesn't seem to be related with Webots.

I've tried foxy with webots on windows and got some error, but don't know if it is matter.

Summary: 27 packages finished [46.7s]
  11 packages had stderr output: webots_ros2 webots_ros2_control webots_ros2_core webots_ros2_epuck webots_ros2_importer webots_ros2_mavic webots_ros2_tesla webots_ros2_tests webots_ros2_tiago webots_ros2_turtlebot webots_ros2_universal_robot

**--- stderr: webots_ros2_control**
CMake Warning at C:/dev_ws/install/webots_ros2_driver/share/webots_ros2_driver/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'webots_ros2_driver' exports library 'Controller.dll' which
  couldn't be found
Call Stack (most recent call first):
  C:/dev_ws/install/webots_ros2_driver/share/webots_ros2_driver/cmake/webots_ros2_driverConfig.cmake:41 (include)
  CMakeLists.txt:36 (find_package)

CMake Warning at C:/dev_ws/install/webots_ros2_driver/share/webots_ros2_driver/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'webots_ros2_driver' exports library 'car.dll' which couldn't be
  found
Call Stack (most recent call first):
  C:/dev_ws/install/webots_ros2_driver/share/webots_ros2_driver/cmake/webots_ros2_driverConfig.cmake:41 (include)
  CMakeLists.txt:36 (find_package)

CMake Warning at C:/dev_ws/install/webots_ros2_driver/share/webots_ros2_driver/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'webots_ros2_driver' exports library 'driver.dll' which couldn't be
  found
Call Stack (most recent call first):
  C:/dev_ws/install/webots_ros2_driver/share/webots_ros2_driver/cmake/webots_ros2_driverConfig.cmake:41 (include)
  CMakeLists.txt:36 (find_package)

**AND the same warning**

setup.py install is deprecated. Use build and pip and other standards-based tools.

dmorris0, 2022-08-21:
@wangruoyao, I got the same error with my Humble install, so I was going to try Foxy too.  Did you get it to work with Foxy despite those errors?

wangruoyao, 2022-08-22:
> @wangruoyao, I got the same error with my Humble install, so I was going to try Foxy too. Did you get it to work with Foxy despite those errors?

Yes, it worked, although I don't know what the errors affect. Anyway, it worked.

dmorris0, 2022-08-22:
I built it with Foxy and the only outputs to stderr were warnings, not those missing DLLs.  Then I was able to control the robot with ROS topics.

wangruoyao, 2022-08-25:
> I built it with Foxy and the only outputs to stderr were warnings, not those missing DLLs. Then I was able to control the robot with ROS topics.

could you double check your output or rebuild it by colcon and focus on the line of stderr: webots_ros2_control

ygoumaz, 2023-02-20:
I am closing this issue for now as we switched to WSL support. Feel free to reopen if the same issue is still occurring in a WSL environment.

closed 2023-02-20 by ygoumaz