cyberbotics/webots_ros2 #461: webots 2022b + webots_ros2 + Ubuntu 22 + ROS2 Humble

royito55, 2022-08-06:
**Is your feature request related to a problem? Please describe.**

I want to start working with webots and ROS2 Humble. Humble only works in Ubuntu 22, and it seems like the way things are going, I want to keep testing webots 2022b. I have already seen that I can install the Humble package from binaries (`apt install ros-humble-webots-ros2`) but it seems like there are issues with the example packages.

When I launch `webots_ros2_turtlebot robot_launch.py`, for example, I get this message in the console:

```
ERROR: Please adapt your project to R2022b following these instructions: https://github.com/cyberbotics/webots/wiki/How-to-adapt-your-world-or-PROTO-to-Webots-R2022b
ERROR: '/opt/ros/humble/share/webots_ros2_turtlebot/worlds/turtlebot3_burger_example.wbt' must be converted because EXTERNPROTO declarations are missing.
INFO: 'TurtleBot3Burger' extern controller: waiting for connection on ipc://1234/TurtleBot3Burger or on tcp://<ip_address>:1234/TurtleBot3Burger
```
and in the terminal, the controller spawner node dies:

```
[spawner-2] [ERROR] [1659808744.228981458] [spawner_diffdrive_controller]: Controller manager not available
[spawner-1] [ERROR] [1659808744.241016900] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner-2]: process has died [pid 10986, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner diffdrive_controller --controller-manager-timeout 50 --ros-args'].
[ERROR] [spawner-1]: process has died [pid 10984, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager-timeout 50 --ros-args'].
```
I assume this is due to the new way extern controllers are handled in 2022b, explained here: https://cyberbotics.com/doc/guide/running-extern-robot-controllers?version=develop

**Describe the solution you'd like**
Updating/adding to the getting started tutorials for them to work in ROS2 Humble and 2022b: https://github.com/cyberbotics/webots_ros2/wiki/Getting-Started

* How to add an extern controller in ROS2 Humble?
* Humble binary packages for Humble to work out of the box (adapting projects to 2022b, as the console error mentioned above suggests)

lukicdarkoo, 2022-09-02:
The libController should be synced in the webots_ros2 develop branch:
https://github.com/cyberbotics/webots_ros2/tree/develop/webots_ros2_driver

It's been a while since the last sync. The libController and Webots are probably not compatible anymore.

sebastianohl, 2022-09-02:
@lukicdarkoo does that mean it should work now or it is still broken an libController need to be fixed?

AlucardLO, 2022-10-21:
I've run into the same issue as described above.

After following the tutorial on [Ros2 Robot Simulation (Webots)](https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Webots.html), I had to rebuild the entire project in 2022b and after doing so was met by the same message.

@lukicdarkoo I see the changes in development, can I assume the changes work and will allow me to run an extern controller?

ygoumaz, 2022-10-21:
`webots_ros2`  2022.1.0 and above versions compatible with Webots R2022b should be available soon. Currently you can use the package on Humble with Webots R2022b by building the package from sources.

ygoumaz, 2022-12-29:
I am closing this issue for now. Feel free to reopen if necessary.

closed 2022-12-29 by ygoumaz