cyberbotics/webots_ros2 #464: How can I get robot's ground truth pose from webots?

bigbigpark, 2022-09-02:
I have successfully built and run [webots_ros2](https://github.com/cyberbotics/webots_ros2).

When I type this command,`ros2 launch webots_ros2_mavic robot_launch.py`
[image]

It worked well.
Now I want to get **Ground Truth Pose** of the Quadrotor as a format of 9-DoF.
I know the position(x, y, z) is got from GPS sensor and the linear acceleration & angular velocity from Inertial Measurement Units.

To see what's topic is streaming, `ros2 topic list`,
[image]

It gives me a correct GPS data, but cannot offer **linear_acceleration**..
How can I get this?

omichel, 2022-09-02:
This is likely due to the fact that the [Mavic2](https://github.com/cyberbotics/webots/blob/71c53a91fab8eab737495c9471b11634d177f781/projects/robots/dji/mavic/protos/Mavic2Pro.proto#L61-L68) robot misses an [Accelerometer](https://www.cyberbotics.com/doc/reference/accelerometer) node. It has a [Compass](https://www.cyberbotics.com/doc/reference/compass), [Gyro](https://www.cyberbotics.com/doc/reference/gyro), [InertialUnit](https://www.cyberbotics.com/doc/reference/inertialunit) (providing only yaw, pitch, roll) and GPS. Adding an Accelerometer to the PROTO file should resolve the issue.

bigbigpark, 2022-09-03:
@omichel 
Thanks for replying.
I'm pretty new about webots. Could you give me some guidance?

omichel, 2022-09-06:
I would recommend you to follow this tutorial: https://docs.ros.org/en/galactic/Tutorials/Advanced/Simulators/Webots.html

closed 2022-09-15 by bigbigpark