cyberbotics/webots_ros2 #465: QSslSocket: cannot resolve EVP_PKEY_base_id

matthew-william-lock, 2022-09-06:
**Describe the Bug**

Bit of a weird one, I am unable to download the textures due to the following error:

```
Warning: QSslSocket: cannot resolve EVP_PKEY_base_id
Warning: QSslSocket: cannot resolve SSL_get_peer_certificate
Warning: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
Warning: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
Warning: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
```

**Steps to Reproduce**
Simply running webots
```
./webots
```

[image].

**System**
 - Webots Version: R2022a
 - ROS Version: ROS2 humble
 - Operating System: Ubuntu 22.04 (Running within Windows 11 WSL ... )
 - Graphics Card: NVIDIA GeForce GTX 1650 Ti

ad-daniel, 2022-09-06:
In order to use R2022a with ubuntu 22 you need to patch OpenSSL.
To do this, download [libssl_1.1.tar.xz](https://cyberbotics.com/files/repository/dependencies/linux64/release/libssl_1.1.tar.xz) and uncompress it with `tar xvf libssl_1.1.tar.xz`, and move its contents to the Webots library folder: `sudo mv openssl-1.1/* /usr/local/webots/lib/webots/`

matthew-william-lock, 2022-09-06:
Thank you for the incredibly quick response, this indeed gets passed the issue!

matthew-william-lock, 2022-09-06:
Although when downloading the Dji example now, the download seems to get stuck at 6%

closed 2022-09-06 by matthew-william-lock