cyberbotics/webots_ros2 #467: Diffdrive_controller and joint_statebroadcaster doesn't open also the map in rviz2 doesn't load.

aayush11101998, 2022-09-16:
Greetings team,

I was trying to simulate multiple tiago robots in webots but I get the following msgs in my log:

**[driver-11] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 26)
[driver-10] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 26)
[driver-11] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 27)
[driver-10] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 27)
[spawner_diffdrive_controller]: Waiting for '/tiago_iron/controller_manager' node to exist
[spawner.py-1] [INFO] [1663171763.423953558] [spawner_joint_state_broadcaster]: Waiting for '/tiago_iron/controller_manager' node to exist
[spawner.py-4] [INFO] [1663171763.424695767] [spawner_diffdrive_controller]: Waiting for '/tiago_iron1/controller_manager' node to exist
[spawner.py-2] [INFO] [1663171763.435691553] [spawner_joint_state_broadcaster]: Waiting for '/tiago_iron1/controller_manager' node to exist.**

I am sharing with you my git repository with this mail so that it would be easier for you to look if there is anything that I am missing, but it looks to me that there is a problem with the map not being received.

https://github.com/aayush11101998/webots_multirobots.git.
you can just run the command ./run.sh after cloning the repository.

Also I am sharing you with some images of my logs and rviz

**Expected behavior:**

Once it starts I am getting the above log messages and in my rviz the global_costmap, downsampled_costmap and local_costmap shows the warning map not recieved.

I suspect that it doesn't  recognise the base_link and base_footprint of my tiagobots.

**Affected Packages**
List of affected packages:
  - webots_ros2_A
  - webots_ros2_B

**Screenshots**
Screenshots are also attached with this problem.
[image]
[image]
[image]
[image]

System:
 - Webots Version: R2022b
 - ROS Version: ROS-GALACTIC 
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card:  NVIDIA GeForce RTX 3060, 6gb
 - Processor: AMD RADEON 7 (5000 SERIES)

aayush11101998, 2022-09-19:
Hi please someone look into this issue.

aayush11101998, 2022-09-23:
@ad-daniel Hi there! Sorry for being impatient butareth3re any updates on the issue? 
Thanks

ad-daniel, 2022-09-23:
Hi, sorry for the delay. There's a few things wrong that I see at first glance. From R2022b, the way extern controllers communicate with Webots has changed (in order to support remote controllers), which is why it doesn't seem to be able to in your project. The warnings as well are related to the fact you're opening a R2022a world with R2022b. If you install R2022a most of the issues should go away (hopefully), if instead you need/want to use R2022b then you'll need to do some adjustments. 
1. I suggest converting the world to this version using the script mentioned [here](https://github.com/cyberbotics/webots/wiki/How-to-adapt-your-world-or-PROTO-to-Webots-R2022b) to get rid of the warnings.
2. Change the environmental variables from `WEBOTS_ROBOT_NAME` to `WEBOTS_CONTROLLER_URL`.
3. You will additionally need to install webots_ros2 from source following [this](https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#building-webots_ros2-package) procedure (master branch) since no official release is available yet for galactic.

Could you try to do these adjustments and try again?

aayush11101998, 2022-09-23:
Thanks a lot @ad-daniel for your reply i will try and let you know.

aayush11101998, 2022-09-29:
Hi! @ad-daniel I tried the above suggestions but still getting the same errors.
**[spawner.py-1] [INFO] [1664459098.603104264] [spawner_joint_state_broadcaster]: Waiting for '/tiago_iron/controller_manager' node to exist
[spawner.py-2] [INFO] [1664459098.603802903] [spawner_joint_state_broadcaster]: Waiting for '/tiago_iron1/controller_manager' node to exist
[spawner.py-3] [INFO] [1664459098.580744546] [spawner_diffdrive_controller]: Waiting for '/tiago_iron/controller_manager' node to exist
[spawner.py-4] [INFO] [1664459098.602769926] [spawner_diffdrive_controller]: Waiting for '/tiago_iron1/controller_manager' node to exist**

Also in my RVIZ the global, downsampled and local costmaps show me warnings "**NO MAPS RECEIVED**".

aayush11101998, 2022-10-05:
Hi! @ad-daniel  is there any update on this issue?

ygoumaz, 2023-02-17:
Hi @aayush11101998, before closing this issue, I want to make sure you were able to run your project. You can now try with Webots R2023a and `2023.0.1` version of the package. If you build the packages from sources, I would even suggest to try from the master branch. Lots of problems with connections and RViz have been solved in past months.

aayush11101998, 2023-02-17:
Thanks you can close it

closed 2023-02-17 by aayush11101998