cyberbotics/webots_ros2 #468: Publishing Twist results in discontinuous movement

matthew-william-lock, 2022-09-19:
**Describe the Bug**

I am publishing a twist message (just 1 message) and it results in the robot only moving for a fraction of a second.

**Steps to Reproduce**
1. Launch the robot
```
ros2 launch webots_ros2_epuck robot_launch.py
```

2. Send one message
```
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "{linear:{x: 2.0}}"
```

**Expected behavior**
I expect the robot to move continuously in the specified direction.

**System**
 - Webots Version: R2022a
 - ROS Version: hubmle
 - Operating System: Ubuntu 22.04

matthew-william-lock, 2022-09-20:
Any ideas on what might be causing this? :eyes:

matthew-william-lock, 2022-09-21:
I found that the defined ```diffdrive_controller``` does not set a [cmd_vel_timeout](http://wiki.ros.org/diff_drive_controller).

Seems that there is obviously some predefined default which is sending a zero speed command to the wheels.

ad-daniel, 2022-09-26:
If you send the command only once then after the timeout kicks indeed a 0 command is automatically sent to the motors. By default this timeout is set to 0.5s, hence the behavior you observed. For the e-puck example to be consistent with the behavior we'd have using a non-ros controllers, I agree we should change it.
Rather than setting an arbitrary timeout, it might be better to set `cmd_vel_timeout : 0.0` so as to disable the behavior entirely.
May I ask you to open a pull request with the proposed change?

matthew-william-lock, 2022-09-26:
@omichel @ad-daniel Just bringing this [discussion](https://github.com/matthew-william-lock/webots_ros2/commit/cf132ab8698e564a1b225d909d7c137ed8ea5ced) here.

I have attempted a zero-valued timeout as @ad-daniel suggested but it does not seem to address the issue. In fact, The robot moves and will almost immediately stop. Looking at the [controller documentation](http://wiki.ros.org/diff_drive_controller), I don't see any obvious way to disable the timeout

ad-daniel, 2022-09-26:
mmh you're right, I tested it locally and it seemed to work, perhaps I forgot to re-build between tries. I've checked in the ros2_control repo and I don't find any way to disable it either other than setting a large value as you mentioned. Perhaps it's best to ask there if there's a more proper solution

ygoumaz, 2023-03-30:
I am closing this issue for now, as the change should be implemented in https://github.com/ros-controls/ros2_controllers/issues/440. If, in the end, something should be change in webots_ros2 as well, please feel free to reopen.

closed 2023-03-30 by ygoumaz