cyberbotics/webots_ros2 #470: Webots 2022b

matthew-william-lock, 2022-09-20:
Hey!

Today I updated to Webots 2022b and faced the following error

```
[robot_state_publisher-4] [INFO] [1663711220.362484768] [robot_state_publisher]: got segment 
[driver-5] Webots doesn't seems to be ready yet: (retry count 1)
[spawner-2] [INFO] [1663711220.528740855] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-1] [INFO] [1663711220.537657199] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[epuck_node-7] [INFO] [1663711220.608125828] [epuck_node]: Epuck node has been started.
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 2)
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 3)
[spawner-2] [INFO] [1663711222.537321431] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-1] [INFO] [1663711222.544650303] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 4)
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 5)
[spawner-2] [INFO] [1663711224.551320351] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-1] [INFO] [1663711224.557480889] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 6)
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 7)
[spawner-2] [INFO] [1663711226.568096993] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-1] [INFO] [1663711226.576513768] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 8)
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 9)
[spawner-2] [INFO] [1663711228.590348901] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-1] [INFO] [1663711228.594707764] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 10)
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 11)
[spawner-2] [INFO] [1663711230.609423667] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-1] [INFO] [1663711230.613123590] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[driver-5] Cannot open file: /tmp/webots-1234/WEBOTS_SERVER (retry count 12)
```

This was after I ran:

```
ros2 launch webots_ros2_epuck robot_launch.py
```
I am wondering if this has been tested with webots 2022b or if something else has gone wrong?

**System**
 - Webots Version: R2022b
 - ROS Version: hubmle
 - Operating System: Ubuntu 22.04

ad-daniel, 2022-09-21:
Currently for R2022b you need to build from the develop branch of the repo, there's still some issues that need ironing out and we'll release the official package soon

ad-daniel, 2022-09-23:
Just FYI, it should work from master branch too now. Official packages should be available with the next sync. Feel free to re-open this if problems persist.

closed 2022-09-23 by ad-daniel