cyberbotics/webots_ros2 #482: Rethink pre-compiled libController submodule

ygoumaz, 2022-10-11:
Robot windows and other dynamic libraries can't be used with WSL (WSL compatibility introduced in #481), because the `libController` is executed on Linux, while the requested libraries are distributed with Webots (running on Windows in this case). This implies that such libraries should be distributed with the `webots_ros2` package, or at least in the pre-compiled [libController repository](https://github.com/cyberbotics/webots-libcontroller/tree/main).

Moreover, the current [libController repository](https://github.com/cyberbotics/webots-libcontroller/tree/main) is containing binaries, which is not a very clean solution.

Here are some discussed solutions:
- Refactor the [libController repository](https://github.com/cyberbotics/webots-libcontroller/tree/main) to make it a lighter but similar architecture as the Webots installation folder,  containing only the necessary compiled libraries.
- Replace the [libController repository](https://github.com/cyberbotics/webots-libcontroller/tree/main) by new nightly-built packages containing the `libController` and other useful libraries. These could be distributed with classic Webots nightly-builds as tarballs for example. `webots_ros2` could download this package to replace the `libController` sub-module.
- Replace the [libController repository](https://github.com/cyberbotics/webots-libcontroller/tree/main) sub-module by the download of the tarball of the last Webots release. All the missing libraries would be present in the `webots_ros2` package. Nightly builds could be used for more up-to-date versions of libraries.

lukicdarkoo, 2022-10-11:
This is a very tough problem to solve. There are two points to consider:
- All files necessary for compiling the ROS packages must be in the Git repository. This is a limitation imposed by the ROS buildfarm. This means that points 2 and 3 you mentioned would probably not be possible.
- Initially, Webots was shipped with the package, but the Webots package got too big for the ROS buildfarm. Given that Webots got way smaller in size since R2021b you could consider shipping Webots as a ROS package. Having Webots as a ROS package would remove the need to distrubute libController. Furthermore, it would be much more convenient for users as Webots would be automatically installed.

ygoumaz, 2022-10-11:
Another solution could be to replace the `libController` submodule with the `webots` repository. The ROS package would compile the necessary files (controllers and libraries). We can also choose to distribute only the necessary files from the repository to avoid having a package too big for the buildfarm.

lukicdarkoo, 2022-10-11:
> Another solution could be to replace the `libController` submodule with the `webots` repository.

The `webots` repository is over 4GB in size, it will likely cause problems in the buildfarm. We should check for a way to distribute Webots as [rosdep](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml). Then, the `webots` package would be a dependency of `webots_ros2`.

ygoumaz, 2022-10-11:
This is also an option. But isn't there a way to specify which files of the sub-module we want to distribute? In this case we could keep only files needed to build the libController, as well as other useful libraries. This would decrease a lot the size of the included files.

ygoumaz, 2022-12-29:
I am closing this issue, as we resolved the problem in #508. The libcontroller sources are now directly commited on this repository and updated with a workflow every night if changes occured in the Webots repository.

closed 2022-12-29 by ygoumaz