cyberbotics/webots_ros2 #486: FAIL to build 'webots_ros2_driver' as 'controller.dll' cannot be found

chaozy, 2022-10-19:
**Describe the Bug**
I tried to build the webots_ros2 with `colcon build --merge-install --executor parallel --continue-on-error` and below is the result:
```
--- stderr: webots_ros2_control
CMake Warning at D:/chaozy/cambridgeStudy/Robotics/ros2_ws/install/share/webots_ros2_driver/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'webots_ros2_driver' exports library 'Controller.dll' which
  couldn't be found
Call Stack (most recent call first):
    ******/webots_ros2_driver/cmake/webots_ros2_driverConfig.cmake:41 (include)
  CMakeLists.txt:41 (find_package)

CMake Warning at   
******/webots_ros2_driver/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'webots_ros2_driver' exports library 'car.dll' which couldn't be
  found
Call Stack (most recent call first):
    ******/webots_ros2_driver/cmake/webots_ros2_driverConfig.cmake:41 (include)
  CMakeLists.txt:41 (find_package)

CMake Warning at   
******/webots_ros2_driver/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'webots_ros2_driver' exports library 'driver.dll' which couldn't be
  found
Call Stack (most recent call first):
  ******/webots_ros2_driver/cmake/webots_ros2_driverConfig.cmake:41 (include)
  CMakeLists.txt:41 (find_package)

---
Finished <<< webots_ros2_control [25.8s]

Summary: 194 packages finished [15min 49s]
  2 packages failed: rosbag2_transport rviz_common
  6 packages had stderr output: rmw_connext_cpp rmw_connext_shared_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp webots_ros2_control webots_ros2_core
  10 packages not processed
```
This error message has been mentioned once in this issue https://github.com/cyberbotics/webots_ros2/issues/458#issuecomment-1207174938. 

There is a solution saying that upgrade the python to 3.10 can solve the problem. But the ros2 foxy has restricted the python version as 3.8 and I don't know what to do with it. 

I am wondering if there is any other solution to this error. 

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**System**
 - Webots Version: 2022aros
 - ROS Version: foxy
 - Operating System: windows10
 - Graphics Card: NVIDIA GeForce RTX 3070

ad-daniel, 2022-10-20:
Hi, a couple of things you should know:
  - native Windows is not supported any longer, if you're on Windows then the suggested approach is to use WSL following these instructions: https://github.com/cyberbotics/webots_ros2/wiki/Complete-Installation-Guide#windows, I think everything is up to date (correct @ygoumaz ?). Native Windows might still work, but will likely be painful to setup.
  - if you need to use R2022a, then you need to target an older version of webots_ros2 (prior to 2022.1.x), since with the latest update the way controllers communicate with webots has changed, it won't work otherwise. To use the latest version of webots_ros2 you should install R2022b instead.
  - version 3.8 is indeed enforced here: https://github.com/cyberbotics/webots_ros2/blob/1e395e1f7c976780bb936bb6b19ae02df81a0759/webots_ros2_driver/CMakeLists.txt#L40 you can try to change it and rebuild to see if it solves your problem

ygoumaz, 2022-10-20:
This is correct. I suggest to use the new WSL (Windows Subsystem for Linux) solution, as it is really simpler and all the ROS part is running under Linux, which is by far more convenient. 

As pointed by @ad-daniel, you should follow these instructions https://github.com/cyberbotics/webots_ros2/wiki/Complete-Installation-Guide#windows. Please note that the WSL compatibility of the package is only available on the `develop` branch. So building from sources is mandatory until we synchronize the new feature. To perform this, you should follow the above instructions and select the `develop` branch when cloning the repository.

closed 2022-10-20 by chaozy