cyberbotics/webots_ros2 #496: webots 2022b + ubuntu 22 + humble:error when launch demo

zhuyeaini9, 2022-10-26:
env:
webots 2022b + ubuntu 22 + humble

build from source as the guide.
build ok.

can open sample world ok.
but when launch demo:ros2 launch webots_ros2_universal_robot multirobot_launch.py
webots log:
**ERROR: Error downloading EXTERNPROTO 'Irb4600-40': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/robots/abb/irb/protos/Irb4600-40.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'LJoint': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/factory/pipes/protos/LJoint.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'HammeredCopper': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/HammeredCopper.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'PipeSection': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/factory/pipes/protos/PipeSection.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'SquareManhole': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/factory/manhole/protos/SquareManhole.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'ElectricalCabinet': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/street_furniture/protos/ElectricalCabinet.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'FireExtinguisher': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/factory/fire_extinguisher/protos/FireExtinguisher.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'CardboardBox': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/factory/containers/protos/CardboardBox.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'DesktopComputer': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/computers/protos/DesktopComputer.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Keyboard': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/computers/protos/Keyboard.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Monitor': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/computers/protos/Monitor.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'OfficeTelephone': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/telephone/protos/OfficeTelephone.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'PlatformCart': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/factory/tools/protos/PlatformCart.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'GlossyPaint': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/GlossyPaint.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Cabinet': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/cabinet/protos/Cabinet.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Desk': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/tables/protos/Desk.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Radiator': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/apartment_structure/protos/Radiator.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'BakelitePlastic': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/BakelitePlastic.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Floor': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/floors/protos/Floor.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'TexturedBackground': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/backgrounds/protos/TexturedBackground.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Robotiq3fGripper': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/devices/robotiq/protos/Robotiq3fGripper.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'RustyMetal': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/RustyMetal.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'ConveyorBelt': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/factory/conveyors/protos/ConveyorBelt.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Can': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/drinks/protos/Can.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Table': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/tables/protos/Table.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'BrushedSteel': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/BrushedSteel.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'OldSteel': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/OldSteel.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'SolidBox': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/solids/protos/SolidBox.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'BrushedAluminium': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/BrushedAluminium.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'FormedConcrete': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/FormedConcrete.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Roughcast': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/Roughcast.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'CorrugatedPlates': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/CorrugatedPlates.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'CorrodedMetal': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/appearances/protos/CorrodedMetal.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'Extrusion': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/geometries/protos/Extrusion.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'WoodenPalletStack': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/factory/pallet/protos/WoodenPalletStack.proto', error code: 99: TLS initialization failed
ERROR: Error downloading EXTERNPROTO 'ConstructionLamp': Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/lights/protos/ConstructionLamp.proto', error code: 99: TLS initialization failed
[... cut ...]
WARNING: WorldInfo > ContactProperties: Cannot download 'https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/default/worlds/sounds/slide.wav', error code: 99: TLS initialization failed
INFO: 'Ros2Supervisor' extern controller: waiting for connection on ipc://1234/Ros2Supervisor or on tcp://<ip_address>:1234/Ros2Supervisor
INFO: 'Ros2Supervisor' extern controller: connected.
INFO: 'UR5e' extern controller: waiting for connection on ipc://1234/UR5e or on tcp://<ip_address>:1234/UR5e
INFO: 'UR5e' extern controller: connected.**

ygoumaz, 2022-11-07:
I am using Webots R2022b on Ubuntu 22.04 with Humble. With the latest version of `webots_ros2` built from sources (2022.1.3), I am unable to reproduce your issue. The problem seems to come from the textures that cannot be downloaded from the web. Could it be related to your internet connection?

There was a problem back then with users on Ubuntu 22 with Webots R2022a that could not download textures either. Are you using Webots R2022b official release or is your Webots also built from sources? For the latter, make sure all your dependencies are completely up-to-date.

I suggest you to update your Webots and `webots_ros2` repositories to the latest commits of master branches. This configuration should work.

zhuyeaini9, 2022-12-08:
finally,I solved it by setup the openssl1.1
my ubuntu default openssl version is 3.0

stefaniapedrazzi, 2023-06-15:
I also get this issue on Ubuntu 22.04 VM with Webots R2023b installed from snap and ROS2 Iron.
If I open the assets links in the browser they are loaded correctly.

Ubuntu 22.04 has OpenSSL 3.0.2 installed on the system by default.
Webots from the snap has libssl1.1.
We should probably add something in the installation documentation to fix the OpenSSL mismatch.

ygoumaz, 2023-07-26:
I tried using the latest snap (R2023b) with the current version of the package. I am on Ubuntu 22.04 with OpenSSL 3.0.2. It works fine. Is there any additional instructions to reproduce the issue?