cyberbotics/webots_ros2 #517: Calculation of `CameraRecognitionObject` seems to be outdated

AKreutz, 2022-11-14:
With [this](https://github.com/cyberbotics/webots/pull/4282) pull request and the release of Webots R2022b, the export of Camera nodes to URDF has been changed such that the coordinate frame of the camera conforms to ROS conventions. The new frame is rotated such that the Z axis points away from the camera. 

I think this means that the code for creating `CameraRecognitionObject` messages needs to be updated:
https://github.com/cyberbotics/webots_ros2/blob/75118d78d545d65bfb90c00f5e275766472fe999/webots_ros2_driver/src/plugins/static/Ros2Camera.cpp#L130-L143
https://github.com/cyberbotics/webots_ros2/blob/75118d78d545d65bfb90c00f5e275766472fe999/webots_ros2_driver/src/plugins/static/Ros2Camera.cpp#L167
The position that is received from Webots is relative to a frame where the X axis points away from the camera. If this position is directly used in ROS, using tf2 to transform recognition objects from the camera frame to the world frame leads to wrong results. Instead, the `position` needs to be transformed to the new frame before publishing it.

omichel, 2022-11-16:
Thank you for raising this issue. Can you contribute a patch for it with a pull request?

AKreutz, 2022-11-17:
Hi, thanks for your response. I opened a pull request with a fix. I added a camera with recognition enabled in the test world. When running this world together with rviz2, the box now shows up at the correct location.

closed 2022-11-24 by AKreutz