cyberbotics/webots_ros2 #522: webots_ros2_driver build failures on ROS 2 Rolling ARM64

nuclearsandwich, 2022-11-16:
**Describe the Bug**

webots_ros2_driver does not currently build on Ubuntu Jammy ARM64

The following additional packages are blocked by this failure
* webots_ros2
* webots_ros2_control
* webots_ros2_epuck
* webots_ros2_mavic
* webots_ros2_tesla
* webots_ros2_tests
* webots_ros2_tiago
* webots_ros2_turtlebot
* webots_ros2_universal_robot

**Links to failing builds**

* https://build.ros2.org/view/Rbin_ujv8_uJv8/job/Rbin_ujv8_uJv8__webots_ros2_driver__ubuntu_jammy_arm64__binary/179/

*Build log excerpt*

```
00:04:37.726 /usr/lib/ccache/c++ -g -O2 -ffile-prefix-map=/tmp/binarydeb/ros-rolling-webots-ros2-driver-2022.1.3=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 CMakeFiles/driver.dir/src/Driver.cpp.o CMakeFiles/driver.dir/src/WebotsNode.cpp.o CMakeFiles/driver.dir/src/PythonPlugin.cpp.o CMakeFiles/driver.dir/src/plugins/Ros2SensorPlugin.cpp.o CMakeFiles/driver.dir/src/plugins/static/Ros2Lidar.cpp.o CMakeFiles/driver.dir/src/plugins/static/Ros2LED.cpp.o CMakeFiles/driver.dir/src/plugins/static/Ros2Camera.cpp.o CMakeFiles/driver.dir/src/plugins/static/Ros2GPS.cpp.o CMakeFiles/driver.dir/src/plugins/static/Ros2RangeFinder.cpp.o CMakeFiles/driver.dir/src/plugins/static/Ros2DistanceSensor.cpp.o CMakeFiles/driver.dir/src/plugins/static/Ros2LightSensor.cpp.o CMakeFiles/driver.dir/src/utils/Math.cpp.o CMakeFiles/driver.dir/src/utils/Utils.cpp.o -o driver   -L/tmp/binarydeb/ros-rolling-webots-ros2-driver-2022.1.3/webots/lib/linux-gnu  -Wl,-rpath,/tmp/binarydeb/ros-rolling-webots-ros2-driver-2022.1.3/webots/lib/linux-gnu:/opt/ros/rolling/lib: /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_generator_py.so /opt/ros/rolling/lib/libstatic_transform_broadcaster_node.so /opt/ros/rolling/lib/libwebots_ros2_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libwebots_ros2_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libwebots_ros2_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libwebots_ros2_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libwebots_ros2_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libwebots_ros2_msgs__rosidl_generator_py.so /usr/lib/aarch64-linux-gnu/libtinyxml2.so -lController -lCppController -ldriver -lCppDriver -lcar -lCppCar /usr/lib/aarch64-linux-gnu/libpython3.10.so /opt/ros/rolling/lib/libclass_loader.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libsensor_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libtf2_ros.so /opt/ros/rolling/lib/libmessage_filters.so /opt/ros/rolling/lib/librclcpp_action.so /opt/ros/rolling/lib/librclcpp.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_generator_py.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/librosgraph_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/liblibstatistics_collector.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_generator_py.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libstatistics_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/librcl_action.so /opt/ros/rolling/lib/librcl.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_generator_py.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_typesupport_c.so /opt/ros/rolling/lib/librcl_interfaces__rosidl_generator_c.so /opt/ros/rolling/lib/librcl_yaml_param_parser.so /opt/ros/rolling/lib/libyaml.so /opt/ros/rolling/lib/libtracetools.so /opt/ros/rolling/lib/librmw_implementation.so /opt/ros/rolling/lib/libament_index_cpp.so /opt/ros/rolling/lib/librcl_logging_spdlog.so /opt/ros/rolling/lib/librcl_logging_interface.so /opt/ros/rolling/lib/libtf2.so /usr/lib/aarch64-linux-gnu/libconsole_bridge.so.1.0 /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_cpp.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_generator_py.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libtf2_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_generator_py.so /opt/ros/rolling/lib/libaction_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libaction_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_generator_py.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_typesupport_c.so /opt/ros/rolling/lib/libunique_identifier_msgs__rosidl_generator_c.so /opt/ros/rolling/lib/libvision_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libstd_msgs__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/librosidl_typesupport_fastrtps_c.so /opt/ros/rolling/lib/libvision_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/librosidl_typesupport_fastrtps_cpp.so /opt/ros/rolling/lib/libfastcdr.so.1.0.24 /opt/ros/rolling/lib/librmw.so /opt/ros/rolling/lib/libvision_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libstd_msgs__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libvision_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libgeometry_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libstd_msgs__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/libbuiltin_interfaces__rosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librosidl_typesupport_introspection_cpp.so /opt/ros/rolling/lib/librosidl_typesupport_introspection_c.so /opt/ros/rolling/lib/libvision_msgs__rosidl_typesupport_cp [... cut ...]
00:04:37.944 /usr/bin/ld: skipping incompatible /tmp/binarydeb/ros-rolling-webots-ros2-driver-2022.1.3/webots/lib/linux-gnu/libCppCar.so when searching for -lCppCar
00:04:37.947 /usr/bin/ld: cannot find -lCppCar: No such file or directory
00:04:37.949 /usr/bin/ld: skipping incompatible /tmp/binarydeb/ros-rolling-webots-ros2-driver-2022.1.3/webots/lib/linux-gnu/libCppCar.so when searching for -lCppCar
00:04:37.951 collect2: error: ld returned 1 exit status
00:04:38.211 make[4]: *** [CMakeFiles/driver.dir/build.make:420: driver] Error 1
```

**System**
 - Operating System: Ubuntu 22.04 (Jammy)
 - Graphics Card: N/A

ygoumaz, 2022-11-16:
As shown in the error messages, there are a few libraries (`libController.so`, etc.) that are pre-compiled on AMD and not compatible with ARM64 architecture. The package has not yet been successfully built on ARM64. However, we are currently working on updating the package to compile these libraries for all architectures and add ARM compatibility.

nuclearsandwich, 2022-11-16:
Thanks, with that information, I'll block these packages on ARM64 for now. When the ARM compatibility is ready, you can open a PR (I'll link the example) to unblock these packages.

closed 2022-12-29 by ygoumaz